platformio/platform-espressif8266 #118: Can't connect to wifi using example code from arduino-webserver

HenneCuppens, 2019-01-19:
Hello everyone,
I am new to platformIO and it seems pretty nice. Only I tried the Arduino-webserver example on my Nodemcu-esp8266, but it is only giving dots in the serial monitor. I know this means it isn't connecting, but I am clueless in why it is not doing so.

The code (allthough, its pretty much the same as example code):
```
#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char* ssid = "I Know You Want Wifi";
const char* password = "15052013";

ESP8266WebServer server(80);

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp8266!");
  digitalWrite(led, 0);
}

void handleNotFound(){
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void){
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void){
  server.handleClient();
}
```

Console output:
```

--- Miniterm on COM9  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

.................................................................................................................................................................................................................................................................................................................................................

```
If anyone knows what I have to change, pls let me know!

HenneCuppens, 2019-01-19:
I checked with a wifiScanner, and I am able to see my SSID, i am getting clueless.

```
19 networks found
1: H368N50B144 (-70)*
2: KPN Fon (-71)
3: UPC249245071 (-67)*
4: VRV9517259696 (-75)*
5: H369AA85979 (-77)*
6: KPN Fon (-76)
7: KPN Fon (-85)
8: H369AC2C0A4 (-86)*
9: Ziggo5108202 (-61)*
10: Ziggo01246 (-84)*
11: Ziggo (-83)*
12: Ziggo84870 (-79)*
13: I know You Want Wifi (-63)*  <--- here it is
14: Ziggo (-64)*
15: TELE2-F3BBC1 (-90)*
16: TELE2-F1A307 (-89)*
17: FRITZ!Box 5490 JB (-87)*
18: H369AAAF509 (-87)*
19: gastfrion (-91)
```

Bolukan, 2019-01-29:
Try to clean the flash memory first: The chip reminds SSID's, maybe something interferes.
esptool.py -p COM5 -b 115200 erase_flash
Replace COM5 with the COM port you use.

ivankravets, 2019-02-18:
You can erase flash with opening PlatformIO IDE Terminal and typing:
```
pio run -t erase
```

closed 2019-02-18 by ivankravets