platformio/platform-espressif8266 #123: " toolchain-xtensa" for "linux_aarch64"

RamilRamizugly, 2019-02-02:
PIO Core Call Error: "PlatformManager: Installing espressif8266\nDownloading\nDownloading\nUnpacking\nUnpacking\nespressif8266 @ 1.8.0 has been successfully installed!\nPackageManager: Installing toolchain-xtensa @ ~1.40802.0\n\nError: Could not find a version that satisfies the requirement '~1.40802.0' for your system 'linux_aarch64'"

ivankravets, 2019-02-02:
Could you try to unpack this https://bintray.com/platformio/dl-packages/download_file?file_path=toolchain-xtensa-linux_armv6l-1.40802.0.tar.gz file to `~/.platformio/packages/toolchain-xtensa`. `package.json` should be located in `~/.platformio/packages/toolchain-xtensa` folder.

Does it work?

SlowmoDK, 2019-02-12:
I have same issue, and after extracting archive to ~/.platformio/packages/toolchain-xtensa
the rest installs (esptool,framework8266 and scons, but now i now get :
sh: 1: xtensa-lx106-elf-g++: not found
*** [.pioenvs/sonoff1/liba92/FastLED_ID126/FastLED.cpp.o] Error 127

ivankravets, 2019-02-18:
The new toolchain has been deployed to our CDN. Please re-run `pio update` command.

closed 2019-02-18 by ivankravets