platformio/platform-espressif8266 #124: ESP8266 Arduino 2.5 - New Release

porkyneal, 2019-02-07:
Now that [2.5.0](https://github.com/esp8266/Arduino/releases) has been released, are we able to get a new release on platformio?

I have loved using the latest GIT for development. But now 2.5.0 is done, It is a great place to put a mark in the sand for my projects, especially for production code.

lucasromeiro, 2019-02-08:
Hi, I'm interested too.

debsahu, 2019-02-08:
@1.9.0 please!

sentoxo, 2019-02-09:
I'm waiting too.

aderusha, 2019-02-16:
Anyone have any advice for adding this manually into PIO until they officially support the 2.5.0 release?

Jason2866, 2019-02-16:
@aderusha 
you could use my repo. I have this done to use core 2.5.0 for the project "Tasmota"
It is not restricted or crippled in any way. Feel free to use until official is released
To use add in platformio.ini the entry:
`platform                  = https://github.com/Jason2866/platform-espressif8266.git#Tasmota`

lucasromeiro, 2019-02-16:
> @aderusha
> you could use my repo. I have this done to use core 2.5.0 for the project "Tasmota"
> It is not restricted or crippled in any way. Feel free to use until official is released
> To use add in platformio.ini the entry:
> `platform = https://github.com/Jason2866/platform-espressif8266.git#Tasmota`

nice...

ivankravets, 2019-02-18:
Please re-test with the latest upstream version https://docs.platformio.org/en/latest/platforms/espressif8266.html#upstream

if it works, we will release in a few next days.

Thanks!

closed 2019-02-18 by ivankravets

aderusha, 2019-02-18:
So far so good over here!  Have to change some code but that's due to changes in 2.5.0.  PIO is working as expected.

FWdeveloper, 2019-02-19:
Unfortunately got FS flash error:

> Building SPIFFS image from ‘data’ directory to .pioenvs\esp_wroom_02\spiffs.bin
> *** [.pioenvs\esp_wroom_02\spiffs.bin] Error 148 

If I go back to `platform = espressif8266` FS image update works well.

debsahu, 2019-02-19:
Default partition has no SPIFFs, one has to use `-Teagle.flash.4m3m.ld` as a **build_flags**

Jason2866, 2019-02-22:
Release generates buggy code. See #128

ivankravets, 2019-02-22:
What do you mean under “buggy code”? Does it work with Arduino IDE? Please share a test project to reproduce your issue.

Jason2866, 2019-02-22:
Project Tasmota does fail
https://github.com/arendst/Sonoff-Tasmota

Jason2866, 2019-02-23:
Solution for bug fix is in #128

commit 48c8b05d (2019-02-18): Update Arduino core to 2.5.0 // Resolve #124