platformio/platform-espressif8266 #125: Support for ESP-IDF style ESP8266_RTOS_SDK

srini1948, 2019-02-18:
The current ESP8266 on PlatformIO is very old. Could you please consider adding the new ESP-IDF style ESP8266_RTOS_SDK. The SDK structure is similar to that of ESP32-IDF.

ivankravets, 2019-03-04:
@WallaceWilliam could you help us with PR? I see you are working on https://github.com/WallaceWilliam/framework-esp8266-rtos-sdk-idf-platformio

No need to rename framework to `framework-esp8266-rtos-sdk-idf`. We will use it instead of old implementation.

WallaceWilliam, 2019-03-04:
this is my hobby. I am ready to do everything in my capabilities.

srini1948, 2019-03-04:
Yes – it will be better to replace the old ESP8266_RTOS_SDK rather than have this custom version. 

Thanks.

From: WallaceWilliam
Sent: Monday, March 4, 2019 6:34 AM
To: platformio/platform-espressif8266
Cc: srini1948; Author
Subject: Re: [platformio/platform-espressif8266] Support for ESP-IDF styleESP8266_RTOS_SDK (#125)

this is my hobby. I am ready to do everything in my capabilities.

WallaceWilliam, 2019-03-04:
version ESP8266_RTOS_SDK in https://github.com/platformio/platform-espressif8266 is 1.5.0-beta
version ESP8266_RTOS_SDK in https://github.com/WallaceWilliam/framework-esp8266-rtos-sdk-idf-platformio is current branch (v3.2.dev-...) of https://github.com/espressif/ESP8266_RTOS_SDK
These versions are not compatible. and all projects written on the old version (1.5.0) will not work.
I have projects in both versions.
for you I can make a separate branch where the framework version 3.2.0 will be framework-esp8266-rtos-sdk. not framework-esp8266-rtos-sdk-idf

srini1948, 2019-03-04:
I am still having problems with the “custom8266” version – it already has the “option” true setting in platform.json.

Once I install this custom platform the new project Wizard of PlatformIO in Visual Studio Code shows Arduino as the only option for ESP8266 boards! 

It seems to remove the three other options: Simba, RTOS and NORTOS

It does not show the newly installed Custom platform at all. I guess a fake Arduino project has to be created and its Platform.ini has to be manually edited.

I don’t need the 1.5 version. It will be great if as you suggest framework-esp8266-rtos-sdk will itself be replaced by the 3.2.0

Thanks.

From: WallaceWilliam
Sent: Monday, March 4, 2019 7:21 AM
To: platformio/platform-espressif8266
Cc: srini1948; Author
Subject: Re: [platformio/platform-espressif8266] Support for ESP-IDF styleESP8266_RTOS_SDK (#125)

version ESP8266_RTOS_SDK in https://github.com/platformio/platform-espressif8266 is 1.5.0-beta
version ESP8266_RTOS_SDK in https://github.com/WallaceWilliam/framework-esp8266-rtos-sdk-idf-platformio is current branch (v3.2.dev-...) of https://github.com/espressif/ESP8266_RTOS_SDK
These versions are not compatible. and all projects written on the old version (1.5.0) will not work.
I have projects in both versions.
for you I can make a separate branch where the framework version 3.2.0 will be framework-esp8266-rtos-sdk. not framework-esp8266-rtos-sdk-idf

ivankravets, 2019-03-04:
@WallaceWilliam 
> These versions are not compatible. and all projects written on the old version (1.5.0) will not work.

You can control the framework version using `platform = espressif8266@x.y.z`.

WallaceWilliam, 2019-03-04:
I rename framework from esp8266-rtos-sdk-idf to esp8266-rtos-sdk
please test

ivankravets, 2019-03-04:
Could you provide PR => https://github.com/platformio/platform-espressif8266/pulls?

srini1948, 2019-03-04:
While trying to install Wallace’s latest version I get an error saying unknown package “tool-esptoolpy”

From: Ivan Kravets
Sent: Monday, March 4, 2019 1:25 PM
To: platformio/platform-espressif8266
Cc: srini1948; Author
Subject: Re: [platformio/platform-espressif8266] Support for ESP-IDF styleESP8266_RTOS_SDK (#125)

Could you provide PR => https://github.com/platformio/platform-espressif8266/pulls?

WallaceWilliam, 2019-03-04:
> While trying to install Wallace’s latest version I get an error saying unknown package “tool-esptoolpy” 
run platformio platform update
and trying again to install

srini1948, 2019-03-04:
Thanks – it worked finally!

From: WallaceWilliam
Sent: Monday, March 4, 2019 3:53 PM
To: platformio/platform-espressif8266
Cc: srini1948; Author
Subject: Re: [platformio/platform-espressif8266] Support for ESP-IDF styleESP8266_RTOS_SDK (#125)

While trying to install Wallace’s latest version I get an error saying unknown package “tool-esptoolpy”
run platformio platform update
and trying again to install

gerritv, 2019-03-19:
I tried (because I really really want to develop something with FreeRTOS) to install per above notes with no success.
I used the Master github from @WallaceWilliam , extracted into Platforms/custom8266, ran 
`pio platforms install custom8266`
 and then 
`pio run`
. Platform.ini has `platform = custom8266` and `framework = esp8266-rtos-sdk`. The result is: (there were no other errors)

`Processing nodemcuv2 (framework: esp8266-rtos-sdk; platform: custom8266; board: nodemcuv2)
-----------------------------------------------------------------------------------------------------------------------
PackageManager: Installing framework-esp8266-rtos-sdk
git version 2.20.1.windows.1
Cloning into 'C:\Users\Gerrit\.platformio\packages\_tmp_installing-9gfs6b-package'...
remote: Enumerating objects: 3395, done.
remote: Counting objects: 100% (3395/3395), done.
remote: Compressing objects: 100% (2734/2734), done.
remote: Total 3395 (delta 552), reused 2622 (delta 488), pack-reused 0
Receiving objects: 100% (3395/3395), 16.97 MiB | 1.12 MiB/s, done.
Resolving deltas: 100% (552/552), done.
Checking out files: 100% (2876/2876), done.
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/custom8266/nodemcuv2.html
PLATFORM: custom Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
TypeError: coercing to Unicode: need string or buffer, NoneType found:
File "C:\python27\lib\site-packages\platformio\builder\main.py", line 169:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\Gerrit\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\Gerrit\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\Users\Gerrit\.platformio\platforms\custom8266\builder\main.py", line 226:
target_elf = env.BuildProgram()
File "C:\Users\Gerrit\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\python27\lib\site-packages\platformio\builder\tools\platformio.py", line 107:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "C:\Users\Gerrit\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\python27\lib\site-packages\platformio\builder\tools\platformio.py", line 319:
SConscript(env.GetFrameworkScript(f), exports="env")
File "C:\Users\Gerrit\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 604:
return method(*args, **kw)
File "C:\Users\Gerrit\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\Gerrit\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\Users\Gerrit\.platformio\platforms\custom8266\builder\frameworks\esp8266-rtos-sdk.py", line 631:
PARTITIONS_TABLE_CSV=full_partitions_csv if isfile(full_partitions_csv) else abspath(partitions_csv))
File "c:\python27\lib\genericpath.py", line 37:
st = os.stat(path) `

WallaceWilliam, 2019-03-20:
edit File "C:\Users\Gerrit.platformio\platforms\custom8266\builder\frameworks\esp8266-rtos-sdk.py"
write before line 630 
print('(full_partitions_csv=', full_partitions_csv)
print('partitions_csv=', partitions_csv)

gerritv, 2019-03-20:
That results in
`
('(full_partitions_csv=', None)
('partitions_csv=', 'partitions.csv')
`

WallaceWilliam, 2019-03-20:
in the file "C:\Users\Gerrit.platformio\platforms\custom8266\builder\frameworks\esp8266-rtos-sdk.py"
line 628
partitions_csv = env.BoardConfig (). get ("build.partitions", partitions_csv)
change to
partitions_csv = 'partitions_singleapp.csv'
This file must be in {framework path}/components/partition_table/

gerritv, 2019-03-20:
That now gives:
`
KeyError: 'LIB_IGNORE':
File "C:\python27\lib\site-packages\platformio\builder\main.py", line 170:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\Gerrit\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\Gerrit\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\Users\Gerrit\.platformio\platforms\custom8266\builder\main.py", line 226:
target_elf = env.BuildProgram()
File "C:\Users\Gerrit\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\python27\lib\site-packages\platformio\builder\tools\platformio.py", line 107:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "C:\Users\Gerrit\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\python27\lib\site-packages\platformio\builder\tools\platformio.py", line 319:
SConscript(env.GetFrameworkScript(f), exports="env")
File "C:\Users\Gerrit\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 604:
return method(*args, **kw)
File "C:\Users\Gerrit\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\Gerrit\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\Users\Gerrit\.platformio\platforms\custom8266\builder\frameworks\esp8266-rtos-sdk.py", line 720:
if d in env['LIB_IGNORE']: continue
File "C:\Users\Gerrit\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 410:
return self._dict[key]
`

gerritv, 2019-03-20:
If I delete the 
`
if d in env['LIB_IGNORE']: continue
`
then build continues until I get
`
xtensa-lx106-elf-gcc: error: Panel\Controller\.pioenvs\nodemcuv2\bootloader.map: No such file or directory
`

WallaceWilliam, 2019-03-20:
1)
please, update paltform: (in terminal)
platformio platform update
compile project.
you will see an error 'Source `partitions.csv' not found, needed by target'
because in sdkconfig.h set 
#define CONFIG_PARTITION_TABLE_CUSTOM_FILENAME "partitions.csv"
or 
#define CONFIG_PARTITION_TABLE_FILENAME "partitions.csv"
fix on "partitions_singleapp.csv"

2)
compile project again
get your compilation error 'bootloader.map: No such file or directory'
delete bootloader.elf in Panel\Controller\.pioenvs\nodemcuv2\
in terminal compile project again with command 'platformio.exe run -v'
you should see a line like this
xtensa-lx106-elf-gcc -nostdlib -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--start-group .....-Tesp8266.bootloader.rom.ld .pioenvs\d1_mini\bootloader\bootloader_start.o -lgcc -lstdc++ -lgcov -Wl,--end-group -Wl,-EL -o .pioenvs\d1_mini\bootloader.elf -Wl,-Map=.pioenvs\d1_mini\bootloader.map

put it here

WallaceWilliam, 2019-03-20:
error in this
"-Wl,-Map=$BUILD_DIR\\bootloader.map"
because there are spaces in the file path this command not work
please, update paltform: (in terminal)
platformio platform update

gerritv, 2019-03-20:
Awesome progress!
It compiled a lot of files, then at the end it said:
`

Linking .pioenvs\8266\firmware.elf
xtensa-lx106-elf-gcc: error: Projects\UII-Panel\Controller\.pioenvs\8266\firmware.map: No such file or directory
`

There is a bootloader.map present now.

WallaceWilliam, 2019-03-20:
please run in terminal
platformio.exe run -v

WallaceWilliam, 2019-03-20:
please, update paltform: (in terminal)
platformio platform update

srini1948, 2019-03-20:
Great work!

gerritv, 2019-03-20:
I did that several times, forgot to tell you. It does nothing (Instead I copy/pasted the .py file that you modified today)
`
Platform custom Espressif 8266
--------
Updating custom8266                      @ 1.0.0          [Detached]
Updating framework-esp8266-rtos-sdk      @ 8378b1a        [Up-to-date]
Updating toolchain-xtensa-lx106-elf-520  @ 5.2.0          [Up-to-date]
`

WallaceWilliam, 2019-03-20:
Updating custom8266 @ 1.0.0 [Detached] - is not good
must be
Platform custom Espressif 8266
Updating custom8266                      @ 8097678        [Up-to-date]

try delete path C:\Users\Gerrit\.platformio\platforms\custom8266
and exec Advanced platform installation from https://github.com/WallaceWilliam/framework-esp8266-rtos-sdk-idf-platformio.git

P.S. after manual change .py file, update platform us run platformio platform update dont work.

gerritv, 2019-03-20:
Ok, that did it!!!!
I uninstalled custom8266, copied new .zip file from your repository, unzipped it in 'packages'.
Then 'pio platform install custom8266' and 'pio run -e 8266' (I have several [env] in my .ini file).

Result is an upload to my device

Thank you very very much. On to developing software using FreeRTOS.

WallaceWilliam, 2019-03-20:
update platform and try again

Platform custom Espressif 8266
Updating custom8266                      @ 8097678        [91a3a1f]

gerritv, 2019-03-20:
Uploaded, yippee.....

Again, thank you soo much for the work and help.

DrywFiltiarn, 2019-04-13:
@WallaceWilliam any clue why PlatformIO on VS Code won't find the tools from the toolchain when trying to "build" code?

`Processing nodemcuv2 (framework: esp8266-rtos-sdk; platform: custom8266; board: nodemcuv2)
---------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing toolchain-xtensa-lx106-elf-520 @ >5.0.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/custom8266/nodemcuv2.html
PLATFORM: custom Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 160MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 1 compatible libraries
Scanning dependencies...
No dependencies
Warning! Unknown upload protocol
Compiling .pioenvs/nodemcuv2/src/main.o
Generating LD script .pioenvs/nodemcuv2/esp8266_out.ld
Generating LD script .pioenvs/nodemcuv2/esp8266_common_out.ld
sh: xtensa-lx106-elf-g++: command not found
sh: xtensa-lx106-elf-gcc: command not found
sh: xtensa-lx106-elf-gcc: command not found
Compiling .pioenvs/nodemcuv2/bootloader/bootloader_start.o
sh: xtensa-lx106-elf-gcc: command not found
Compiling .pioenvs/nodemcuv2/bootloader/esp8266/source/ets_printf.o
sh: xtensa-lx106-elf-gcc: command not found
Compiling .pioenvs/nodemcuv2/bootloader/log/log.o
sh: xtensa-lx106-elf-gcc: command not found`

What strikes me is that each time I try to build it thinks it needs to reinstall the toolchain again. Also when I do a `pio update` I get the following result:

`Platform custom Espressif 8266
--------
Updating custom8266                      @ 1.0.0          [Detached]
Updating framework-esp8266-rtos-sdk      @ 2105cf0        [Up-to-date]
Uninstalling toolchain-xtensa-lx106-elf-520 @ 5.2.0:    [OK]`

DrywFiltiarn, 2019-04-15:
The problem lies in the fact that your implementation so far is only aimed at Windows. As I’m on a Mac it could obviously not install the required tools. I have been working on 2 forks of your repositories to also support the osx toolchain. I will come back on this later.

WallaceWilliam, 2019-04-16:
@DrywFiltiarn I redisign _platform.json_ and add support for OSX
But I can't test it because I don't have OSX
And in _xtensa-lx106-elf\manifest.json_  set "system": ["linux_x86_64"] But it's not right. you must determine the type of systems for OSX
otherwise the toolchain for OSX will not be installed

DrywFiltiarn, 2019-04-16:
@WallaceWilliam the systems for OSX will be: "darwin_x86_64" or "darwin_i386" both use the same toolchain download.

What you configured now for the manifast is invalid.
`
{
  "system": ["linux_x86_64"],
  "url": "https://github.com/WallaceWilliam/xtensa-lx106-elf/raw/master/xtensa-lx106-elf-osx-1.22.0-92-g8facf4c-5.2.0.tar.gz",
  "version": "5.2.0"
}
`

needs to be
`
{
  "system": ["darwin_x64_86", "darwin_i386"],
  "url": "https://github.com/WallaceWilliam/xtensa-lx106-elf/raw/master/xtensa-lx106-elf-osx-1.22.0-92-g8facf4c-5.2.0.tar.gz",
  "version": "5.2.0"
}
`

You could add Linux as well, but that would require you to copy the correct tar.gz from espressif and modifity it to contain the correct package.json.

The manifest in my fork https://github.com/DrywFiltiarn/xtensa-lx106-elf-1.22.0-92-g8facf4c-5.2.0/blob/master/manifest.json is configured correctly for osx.

boriss2oo6, 2019-04-26:
@WallaceWilliam Hi,

After an update on platformio i'm stuck with "fatal error: spiffs.h: No such file or directory"

> Executing task in folder Esp8266: C:\Users\Boriss\.platformio\penv\Scripts\platformio.exe run <

Processing d1_mini (platform: custom8266; board: d1_mini; framework: esp8266-rtos-sdk)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/custom8266/d1_mini.html
PLATFORM: custom Espressif 8266 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Embed file $PROJECT_DIR\src/html/details.html
Embed file $PROJECT_DIR\src/html/header.html
Embed file $PROJECT_DIR\src/html/input.html
Embed file $PROJECT_DIR\src/html/output.html
Embed file $PROJECT_DIR\src/html/console.html
Embed file $PROJECT_DIR\src/html/favicon.ico
Embed file $PROJECT_DIR\src/html/server.pem
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 0 compatible libraries
Scanning dependencies...
No dependencies
Evaluated 1 nodes
Evaluated 2 nodes
Evaluated 3 nodes
Evaluated 4 nodes
Evaluated 5 nodes
Evaluated 6 nodes
Evaluated 7 nodes
Evaluated 8 nodes
Evaluated 9 nodes
Evaluated 10 nodes
Evaluated 11 nodes
Compiling .pioenvs\d1_mini\spiffs\esp_spiffs.o
Compiling .pioenvs\d1_mini\spiffs\spiffs_api.o
C:\users\boriss\.platformio\packages\framework-esp8266-rtos-sdk\components\spiffs\esp_spiffs.c:16:20: fatal error: spiffs.h: No such file or directory

****************************************************************
* Looking for spiffs.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:spiffs.h"
* Web  > https://platformio.org/lib/search?query=header:spiffs.h
*
****************************************************************

compilation terminated.
Evaluated 12 nodes
In file included from C:\users\boriss\.platformio\packages\framework-esp8266-rtos-sdk\components\spiffs\spiffs_api.c:20:0:
C:\users\boriss\.platformio\packages\framework-esp8266-rtos-sdk\components\spiffs\spiffs_api.h:22:20: fatal error: spiffs.h: No such file or directory

****************************************************************
* Looking for spiffs.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:spiffs.h"
* Web  > https://platformio.org/lib/search?query=header:spiffs.h
*
****************************************************************

compilation terminated.
Compiling .pioenvs\d1_mini\libsodium\libsodium\src\libsodium\crypto_core\hsalsa20\ref2\core_hsalsa20_ref2.o
Compiling .pioenvs\d1_mini\libsodium\libsodium\src\libsodium\crypto_core\salsa\ref\core_salsa_ref.o
Compiling .pioenvs\d1_mini\libsodium\libsodium\src\libsodium\crypto_generichash\blake2b\generichash_blake2.o
Compiling .pioenvs\d1_mini\libsodium\libsodium\src\libsodium\crypto_generichash\blake2b\ref\blake2b-compress-avx2.o
Compiling .pioenvs\d1_mini\libsodium\libsodium\src\libsodium\crypto_generichash\blake2b\ref\blake2b-compress-ref.o
*** [.pioenvs\d1_mini\spiffs\esp_spiffs.o] Error 1
Compiling .pioenvs\d1_mini\libsodium\libsodium\src\libsodium\crypto_generichash\blake2b\ref\blake2b-compress-sse41.o*** [.pioenvs\d1_mini\spiffs\spiffs_api.o] Error 1

==================================================================================================== [ERROR] Took 5.92 seconds ====================================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

WallaceWilliam, 2019-04-29:
Hi
add in the sdkconfig.h
#define CONFIG_USING_SPIFFS 1

boriss2oo6, 2019-04-29:
Thanks @WallaceWilliam !

My first solution was to delete spiffs folder from sdk components. The solution provided by you is great but still the project gives me a few compiling errors which i solved by adding to sdkconfig.h:

#define CONFIG_USING_SPIFFS 1
#define CONFIG_SPIFFS_META_LENGTH 4
#define CONFIG_SPIFFS_GC_MAX_RUNS 10
#define CONFIG_SPIFFS_MAX_PARTITIONS 3
#define CONFIG_SPIFFS_OBJ_NAME_LEN 32
#define CONFIG_SPIFFS_PAGE_SIZE 256

SimSonic, 2019-12-08:
Hi all!
Several days ago Espressif has deprecated NonOS SDK.

```
Support Policy for ESP8266 NonOS
Starting from December 2019,

We will not add any new features to the ESP8266 NonOS SDK.
We will only fix critical bugs in the ESP8266 NonOS SDK.
We will only maintain the master branch of ESP8266 NonOS SDK, which is a continuously bug-fix version based on v3.0. This means:
All other released branches will not be updated.
All the future versions will be released from only the master branch mentioned above.
It is suggested that the ESP8266_RTOS_SDK, instead of ESP8266 NonOS SDK, be used for your projects.
The latest ESP8266_RTOS_SDK allows users to develop applications using an architecture that are compatible with the SDKs of all Espressif chips, including ESP8266 series, ESP32 series, and the upcoming new series of chips. Switching to ESP8266_RTOS_SDK will helps users to:

Eliminate the necessity to maintain more than one applications (for different chips), thus greatly reducing maintenance costs.
Easily switch to other Espressif chips in the future for enhanced flexibility, less dependency, and reduced time-to-market.
Thank you for your interest in Espressif products.
```
© https://github.com/espressif/ESP8266_NONOS_SDK

What about updating RTOS in platformio to the up to date version (release/v3.3)?

Can you concentrate your forces in support of a single framework, please?

I'm trying to write wemos d1 mini pro firmware for a couple of days with different approaches.
The only +/- correctly working is VS Code + PIO (a tried also CLion with esp8266 plugin) and other CLI approaches that are too hardcore for me (I'm on Windows).

Thanks.

mriksman, 2020-03-10:
@WallaceWilliam 

I installed the platform using the command
`pio platform install https://github.com/WallaceWilliam/framework-esp8266-rtos-sdk-idf-platformio.git`
Which installed
```
custom8266 @ 0fff40c has been successfully installed!
PackageManager: Installing toolchain-xtensa-lx106-elf @ 5.2.0+100
```

I created a new project, selecting a board from the custom8266 section, and went to compile.

 ```
- toolchain-xtensa-lx106-elf 5.2.0+100
  File "C:\PlatformIO\.platformio\platforms\custom8266\builder\frameworks\esp8266-rtos-sdk.py", line 242
    line = line.replace('"','')
                              ^
TabError: inconsistent use of tabs and spaces in indentation
```
The README goes on to say install CMake, install GNU Make, rename src to main, build project using idf.py…

So - then to build a project, it is no longer using PlatformIO…? Then what was the purpose of the WallaceWilliam Platform?

I’m confused…

WallaceWilliam, 2020-03-11:
Greetings.
the build script is getting harder. with every new version of RTOS the script changes.
I don’t have the time to make changes to maintain platformio, and I don’t see the point of why.
but I'm used to working in VSCode. that's why I made such an opportunity for myself.
I apologize if I did not live up to your expectations.

WallaceWilliam, 2020-03-11:
in fact, vscode is used as a text editor

mriksman, 2020-03-11:
Sorry, I didn't mean to sound ungrateful. I'm just confused as to what your platform provides. 
So from what I can gather, right now it essentially just downloads the SDK and the toolchain. 
But everything else needs to be installed/configured/used based on the ESP-IDF installation instructions (CMake, Ninja, etc).
Are there any other benefit your `platform` provides that I am missing?

WallaceWilliam, 2020-03-11:
the project in its current state has no advantages

maxgerhardt, 2021-01-13:
Maybe a approach like the platform-espressif32 and ESP-IDF build script can be taken -- instead of re-implementing everything in Python, call into the native CMake environment (https://github.com/platformio/platform-espressif32/blob/e542df17d188a4545965ca8bafe6d0882e31d1c0/builder/frameworks/espidf.py#L631-L640). That is very robust against changes. 

@ivankravets This topic of an outtdated ESP8266-RTOS-SDK has also appeared in https://github.com/platformio/platform-espressif32/blob/e542df17d188a4545965ca8bafe6d0882e31d1c0/builder/frameworks/espidf.py#L631-L640 recently

jcw, 2021-03-07:
Maybe this is better discussed on the community forum, but I'd like to try and keep the ESP8266 alive and kicking, if there's not _too_ much work involved. The Non-OS version is deprecated, but the version based on FreeRTOS still has potential, IMO.

The ESP32 world is now CMake-driven, it seems, which adds considerable overhead in PIO (in terms of slowing down the edit-run dev cycle), whereas on ESP8266 it's still a very snappy build and upload (1.843 MBaud works fine here). For me, the 8266 offers a way to develop rapidly, while working on code which should also port well to ESP32. And depending on where you're coming from,  100 kB flash + 40 kB RAM free for application code, while having an RTOS-based TCP/IP is pretty useful.

(For reference: I've been doing most of my recent dev work on STM32, and the 1-sec dev cycles are so effective that I really want to get to comparable cycle times on the other systems. On ESP8266-RTOS it's under 5s, which is still very decent.)

VictorCorchez, 2023-07-13:
any plans for this yet?

hernot, 2025-06-08:
by the way issue 219 requests to upgrade existing v1.4 to the newer v3.x