platformio/platform-espressif8266 #127: Cannot override default ld script after version 2.0

joaoTrevizoli, 2019-02-21:
If i change the flash size by overrinding the ld script with any of the available ones the board keeps wdt reseting (cause:4, boot mode:(1,6)), this problem never happened in previous versions, just now after I updated to version 2.0.

I am using in platformio.ini the following settings:

[env:nodemcuv2_dev]
platform = espressif8266
framework = arduino
board = nodemcuv2
upload_speed = 115200
monitor_speed = 115200
build_flags =
-Wl,-Teagle.flash.2m.ld
lib_deps =
31
19
64
83
576
663
748

I tried a few ld scripts available at:
https://github.com/esp8266/Arduino/tree/master/tools/sdk/ld

none seems to work

ivankravets, 2019-02-23:
Sorry, I see a lot of issues with new toolchain. I've just downgraded to the previous version. Please re-test the upstream version => https://docs.platformio.org/en/latest/platforms/espressif8266.html#stable-and-upstream-versions

Does it work now? If yes, I'll make a hot release.

ivankravets, 2019-02-24:
Resolved in https://github.com/platformio/platform-espressif8266/releases/tag/v2.0.1

Please `PlatformIO IDE > PIO Home > Platforms > Updates` or `$ pio update`.

closed 2019-02-24 by ivankravets

joaoTrevizoli, 2019-02-25:
I done that Ivan, now there is the following ocurring:

Platform Manager
================
Platform Arduino
--------
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/update.py", line 46, in cli
    ctx.invoke(cmd_platform_update, only_check=only_check)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/platform.py", line 385, in platform_update
    platform, only_packages=only_packages, only_check=only_check)
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/platform.py", line 121, in update
    p = PlatformFactory.newPlatform(pkg_dir)
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/platform.py", line 250, in newPlatform
    _instance = platform_cls(join(platform_dir, "platform.json"))
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/platform.py", line 460, in __init__
    self._manifest = util.load_json(manifest_path)
  File "/usr/local/lib/python2.7/dist-packages/platformio/util.py", line 185, in load_json
    with open(file_path, "r") as f:
IOError: [Errno 2] No such file or directory: '/home/joao/.platformio/platforms/Arduino/platform.json'

============================================================

joaoTrevizoli, 2019-02-25:
I were able to solve it by manually removing the folder.  aka rm -rf .platformio/platforms/Arduino