platformio/platform-espressif8266 #128: Release Core 2.5.0 generated code is faulty

Jason2866, 2019-02-22:
Project Tasmota doesnt work with release 2.5.0. doesnt boot
https://github.com/arendst/Sonoff-Tasmota/issues/5301#issuecomment-466312768
With this entry in platformio.ini (self hosted release core 2.5.0) from me it does work
```
; *** Esp8266 core for Arduino version 2.5.0
;platform                  = espressif8266@2.0.0
platform                  = https://github.com/Jason2866/platform-espressif8266.git#Tasmota 
```

mcspr, 2019-02-23:
Yep, I too can reproduce endless wdt resets with my testing branch for Espurna xoseperez/espurna#1559 and with Sonoff-Tasmota 2.5.0 development build. BUT I could not reproduce with basic examples.
I tried to compare build flags - only notable difference is that basic pio project prepends board ldscript (`-T eagle.flash.1m.ld`), but in Tasmota build the board ldscript is ignored and custom one added later in argument list via `-Wl,-Teagle.flash.1m.ld`

Because platformio-core handles ldscripts initially, I tried adding reordering them:
https://github.com/mcspr/platformio-core/commit/708eff208877cefbe7d05805bc65ab975f723e3b
(also renaming -Wl,-T to -T, but idk if that matters)

That fixed the builds for me 🤷‍♂️
@Jason2866 can you try that patch?

Jason2866, 2019-02-23:
Your changes do work. Had to do some small changes to get it working: Here is the file platformio.py:
```
# Copyright (c) 2014-present PlatformIO <contact@platformio.org>
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#    http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

from __future__ import absolute_import

import re
import sys
from glob import glob
from os import sep, walk
from os.path import basename, dirname, isdir, join, realpath

from SCons import Builder, Util
from SCons.Script import (COMMAND_LINE_TARGETS, AlwaysBuild,
                          DefaultEnvironment, Export, SConscript)

from platformio.util import glob_escape, pioversion_to_intstr

SRC_HEADER_EXT = ["h", "hpp"]
SRC_C_EXT = ["c", "cc", "cpp"]
SRC_BUILD_EXT = SRC_C_EXT + ["S", "spp", "SPP", "sx", "s", "asm", "ASM"]
SRC_FILTER_DEFAULT = ["+<*>", "-<.git%s>" % sep, "-<svn%s>" % sep]
SRC_FILTER_PATTERNS_RE = re.compile(r"(\+|\-)<([^>]+)>")

def scons_patched_match_splitext(path, suffixes=None):
    """Patch SCons Builder, append $OBJSUFFIX to the end of each target"""
    tokens = Util.splitext(path)
    if suffixes and tokens[1] and tokens[1] in suffixes:
        return (path, tokens[1])
    return tokens

def _build_project_deps(env):
    project_lib_builder = env.ConfigureProjectLibBuilder()

    # prepend project libs to the beginning of list
    env.Prepend(LIBS=project_lib_builder.build())
    # prepend extra linker related options from libs
    env.PrependUnique(
        **{
            key: project_lib_builder.env.get(key)
            for key in ("LIBS", "LIBPATH", "LINKFLAGS")
            if project_lib_builder.env.get(key)
        })

    projenv = env.Clone()

    # CPPPATH from dependencies
    projenv.PrependUnique(CPPPATH=project_lib_builder.env.get("CPPPATH"))
    # extra build flags from `platformio.ini`
    projenv.ProcessFlags(env.get("SRC_BUILD_FLAGS"))

    is_test = "__test" in COMMAND_LINE_TARGETS
    if is_test:
        projenv.BuildSources("$BUILDTEST_DIR", "$PROJECTTEST_DIR",
                             "$PIOTEST_SRC_FILTER")
    if not is_test or env.get("TEST_BUILD_PROJECT_SRC") == "true":
        projenv.BuildSources("$BUILDSRC_DIR", "$PROJECTSRC_DIR",
                             env.get("SRC_FILTER"))
[... 267 lines cut ...]
    return env.StaticLibrary(
        env.subst(variant_dir),
        env.CollectBuildFiles(variant_dir, src_dir, src_filter))

def BuildSources(env, variant_dir, src_dir, src_filter=None):
    nodes = env.CollectBuildFiles(variant_dir, src_dir, src_filter)
    DefaultEnvironment().Append(
        PIOBUILDFILES=[env.Object(node) for node in nodes])

def exists(_):
    return True

def generate(env):
    env.AddMethod(BuildProgram)
    env.AddMethod(ParseFlagsExtended)
    env.AddMethod(ProcessFlags)
    env.AddMethod(ProcessUnFlags)
    env.AddMethod(IsFileWithExt)
    env.AddMethod(MatchSourceFiles)
    env.AddMethod(CollectBuildFiles)
    env.AddMethod(BuildFrameworks)
    env.AddMethod(BuildLibrary)
    env.AddMethod(BuildSources)
    return env 
```

mcspr, 2019-02-23:
Thanks. Looks like the pio version is slightly different in git:
```diff
--- platformio_from_issue_128.py
+++ platformio/builder/tools/platformio.py
@@ -24,7 +24,7 @@
 from SCons.Script import (COMMAND_LINE_TARGETS, AlwaysBuild,
                           DefaultEnvironment, Export, SConscript)

-from platformio.util import glob_escape, pioversion_to_intstr
+from platformio.util import glob_escape, pioversion_to_intstr, string_types

 SRC_HEADER_EXT = ["h", "hpp"]
 SRC_C_EXT = ["c", "cc", "cpp"]
@@ -208,7 +208,7 @@
     # provided with a -U option // Issue #191
     undefines = [
         u for u in env.get("CCFLAGS", [])
-        if isinstance(u, basestring) and u.startswith("-U")
+        if isinstance(u, string_types) and u.startswith("-U")
     ]
     if undefines:
         for undef in undefines:
```

ivankravets, 2019-02-23:
Sorry, I see a lot of issues with new toolchain. I've just downgraded to the previous version. Please re-test the upstream version => https://docs.platformio.org/en/latest/platforms/espressif8266.html#stable-and-upstream-versions

Does it work now? If yes, I'll make a hot release.

CRCinAU, 2019-02-24:
So had a quick read, but I'm not sure about PIOs internals - but what I do know is that I can't create a working binary for *any* of my projects at the moment... 

As a summary:
Works:
platform = https://github.com/platformio/platform-espressif8266.git

Fails: 
platform = espressif8266
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage

mcspr, 2019-02-24:
No problems with the `develop` platform

and this was already tested with the [Tasmota fork of the platform](https://github.com/Jason2866/platform-espressif8266/blob/Tasmota/platform.json), since it used old toolchain all along (since the nonos sdk v3 was merged, i think)

CRCinAU, 2019-02-24:
Also - possibly related (although this is way more indepth that I understand the SDKs etc):

https://github.com/esp8266/Arduino/issues/5784

ivankravets, 2019-02-24:
Resolved in https://github.com/platformio/platform-espressif8266/releases/tag/v2.0.1

Please `PlatformIO IDE > PIO Home > Platforms > Updates` or `$ pio update`.

closed 2019-02-24 by ivankravets

CRCinAU, 2019-02-24:
Just to confirm (and for the record when people hit this via Google), this will fix built issues when using:
```
platform = espressif8266
```

Correct?

ivankravets, 2019-02-24:
Yes, but please do `$ pio update` if you have dev/platform already installed. Otherwise, use `platform = espressif8266@>2.0.0`

CRCinAU, 2019-02-24:
Thanks - and again, for the record, a working system should look like the following:

```
$  pio update
Updating tool-scons                      @ 2.20501.7      [Up-to-date]
Platform Manager
================
Platform Espressif 8266
--------
Updating espressif8266                   @ 2.0.1          [Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20500.190223 [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]
```