platformio/platform-espressif8266 #129: Error in package toolchain-xtensa after upgraded to 2.0.0

Super169, 2019-02-22:
I just upgraded from 1.8.0 to 2.0.0, and I cannot build the program anymore.
There are many error related to the package toolchain-xtensa (see the message at the end).
It works fine in 1.8.0, just failed in 2.0.0, and according to the release note of 2.0.0, toolchain-xtensa has been updated, is there anything I has to do with this update?  I have cleaned the project and build again, but still failed.
Is there any way to fallback to 1.8.0?

=======================
Build errors:
=======================
In file included from c:\users\super\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\map:60:0,
from C:\users\super\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266mDNS\src/LEAmDNS.h:112,
from C:\users\super\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266mDNS\src/ESP8266mDNS.h:46,
from lib\SimpleWiFiManager/SimpleWiFiManager.h:8,
from src\robot.h:28,
from E:/SourceTree/PlatformIO/RobotControl/src/RobotControl.ino:99:
c:\users\super\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_tree.h:735:25: error: macro "swap" requires 2 arguments, but only 1 given
swap(_Rb_tree& __t);
^
c:\users\super\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_tree.h:956:59: error: macro "swap" passed 10 arguments, but takes just 2
_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>& __y)
^
c:\users\super\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_tree.h:957:19: error: macro "swap" requires 2 arguments, but only 1 given
{ __x.swap(__y); }
^
c:\users\super\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_tree.h:1262:66: error: macro "swap" passed 5 arguments, but takes just 2
swap(_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>& __t)
^
In file included from c:\users\super\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\map:61:0,
from C:\users\super\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266mDNS\src/LEAmDNS.h:112,
from C:\users\super\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266mDNS\src/ESP8266mDNS.h:46,
from lib\SimpleWiFiManager/SimpleWiFiManager.h:8,
from src\robot.h:28,
from E:/SourceTree/PlatformIO/RobotControl/src/RobotControl.ino:99:
c:\users\super\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_map.h:286:16: error: macro "swap" requires 2 arguments, but only 1 given
this->swap(__x);
^
c:\users\super\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_map.h:780:20: error: macro "swap" requires 2 arguments, but only 1 given
swap(map& __x)
^
c:\users\super\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_map.h:781:27: error: macro "swap" requires 2 arguments, but only 1 given
{ _M_t.swap(__x._M_t); }
^
c:\users\super\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_map.h:1017:40: error: macro "swap" passed 8 arguments, but takes just 2
map<_Key, _Tp, _Compare, _Alloc>& __y)
^
c:\users\super\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_map.h:1018:19: error: macro "swap" requires 2 arguments, but only 1 given
{ __x.swap(__y); }
^
In file included from c:\users\super\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\map:62:0,

Super169, 2019-02-22:
Just fallback to 1.8.0 by download the zip file and replace  .platformio\platforms\espressif8266 folder, everything works fine now.  But I cannot upgrade this platform due to this issue.

May I know how to fix the error in toolchain-xtensa after upgrade to 2.0.0?

ivankravets, 2019-02-23:
Sorry, I see a lot of issues with new toolchain. I've just downgraded to the previous version. Please re-test the upstream version => https://docs.platformio.org/en/latest/platforms/espressif8266.html#stable-and-upstream-versions

Does it work now? If yes, I'll make a hot release.

ivankravets, 2019-02-24:
Resolved in https://github.com/platformio/platform-espressif8266/releases/tag/v2.0.1

Please `PlatformIO IDE > PIO Home > Platforms > Updates` or `$ pio update`.

closed 2019-02-24 by ivankravets

Super169, 2019-02-25:
I just try to build my project https://github.com/Super169/RobotControl/tree/dev  after upgrade 2.0.1.
But unfortunately, the same error reported,  have to fallback to 1.8.0 again.

Super169, 2019-02-26:
The problem has been solved, as one of the old arudino library has defined "swap", which cause issue with EP8266mDNS in 2.0.1, but there has no problem in 1.8.0.

Just remove the define statement and it can be compiled without error.  However, don't why it works in 1.8.0 but not 2.0+.