platformio/platform-espressif8266 #130: My Skecth work on Arduino IDE but in pio not

semos2k, 2019-02-23:
Hi, this sketch work on Arduino IDE:

#include <Arduino.h>
#include <Wire.h>  // Only needed for Arduino 1.6.5 and earlier
#include "BasicStepperDriver.h"
#include "SSD1306Wire.h" 

#define DIR_A  D5
#define STEP_A D6
#define DIR_B  D7
#define STEP_B D8

#define END_STOP_A  D3

#define MOTOR_STEPS 200
#define MICROSTEPS 2
#define RPM 120

SSD1306Wire  display(0x3c, D2, D1);
BasicStepperDriver stepperA(MOTOR_STEPS, DIR_A, STEP_A);
BasicStepperDriver stepperB(MOTOR_STEPS, DIR_B, STEP_B);

//Uncomment line to use enable/disable functionality
//BasicStepperDriver stepper(MOTOR_STEPS, DIR, STEP, SLEEP);

void setup() {
    pinMode(END_STOP_A, INPUT);
  
    display.init();
    display.flipScreenVertically();
    display.setFont(ArialMT_Plain_16);
    display.clear();
    display.drawString(1, 1, String("Iniciando ..."));
    display.display();

    stepperA.begin(RPM, MICROSTEPS);
    stepperB.begin(RPM, MICROSTEPS);
    // if using enable/disable on ENABLE pin (active LOW) instead of SLEEP uncomment next line
    // stepper.setEnableActiveState(LOW);
}

void show(int pos){
    int endstop = digitalRead(END_STOP_A);
    display.clear();
    display.setTextAlignment(TEXT_ALIGN_LEFT);
    display.drawString(1, 1, String(pos));
    display.drawString(1, 16, String(endstop));
    display.display();
}

void rotate(int pos, int rot){
    show(pos);
    stepperA.rotate(rot);
    stepperB.rotate(rot);
    delay(2000);
}

void move(int pos, int movement){
    show(pos);
    stepperA.move(movement);
    stepperB.move(movement);
    delay(2000);
}

void loop() {
    rotate(1, 360);
    rotate(2, 360);
    rotate(3, 360);

    move(4, -MOTOR_STEPS*MICROSTEPS);
    move(5, -MOTOR_STEPS*MICROSTEPS);
    move(6, -MOTOR_STEPS*MICROSTEPS);
}

----------------------------

[platformio]
#src_dir     = Marlin
build_dir   = .pioenvs
lib_dir     = .piolib
libdeps_dir = .piolibdeps
env_default = d1_mini_pro

[common]
build_flags = -Wl,-Teagle.flash.16m15m.ld -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
lib_deps    = 
    StepperDriver@1.2.0
    ESP8266_SSD1306@4.0.0

[env:d1_mini_pro]
platform    = espressif8266
framework   = arduino
board       = d1_mini_pro
lib_deps    = ${common.lib_deps}
build_flags = ${common.build_flags}
upload_speed = 921600
monitor_speed = 115200

thx

Jason2866, 2019-02-23:
Probably same issue as #128

ivankravets, 2019-02-23:
Do you use the same settings in Arduino IDE? LD Script? lwIP profile?

ivankravets, 2019-02-24:
Resolved in https://github.com/platformio/platform-espressif8266/releases/tag/v2.0.1

Please `PlatformIO IDE > PIO Home > Platforms > Updates` or `$ pio update`.

closed 2019-02-24 by ivankravets