platformio/platform-espressif8266 #132: complile with "board=d1_mini_pro" fails

jipp, 2019-02-27:
starting with v2.0.0 and also with 2.0.1 the problem appears.
using 1.8.0 it also works.

I have the following env config:

> [env:d1_mini_pro]
> platform = espressif8266
> board = d1_mini_pro
> framework = arduino
> build_flags = ${global.build_flags}
> extra_scripts = moveBinary.py
> upload_speed = 115200
> monitor_speed = 115200

the compile stops with the following error message:

> Current build targets ['buildprog', 'size']
> Error: Could not find 'eagle.flash.16m.ld' LD script in LDPATH 'c:\Users\eedwoke\Documents\GitHub\mqttSensorApp\.pioenvs\d1_mini_pro c:\Users\eedwoke\Documents\GitHub\mqttSensorApp\.pioenvs\d1_mini_pro\ld C:\Users\eedwoke\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lib C:\Users\eedwoke\.platformio\packages\framework-arduinoespressif8266\tools\sdk\ld C:\Users\eedwoke\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\lib'
> Error: Could not find 'eagle.flash.16m.ld' LD script in LDPATH 'c:\Users\eedwoke\Documents\GitHub\mqttSensorApp\.pioenvs\d1_mini_pro c:\Users\eedwoke\Documents\GitHub\mqttSensorApp\.pioenvs\d1_mini_pro\ld C:\Users\eedwoke\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lib C:\Users\eedwoke\.platformio\packages\framework-arduinoespressif8266\tools\sdk\ld C:\Users\eedwoke\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\lib'
> *** [.pioenvs\d1_mini_pro\firmware.bin] Error 1

replacing the "d1_mini_pro" with "d1_mini" and using this env it works fine:

> [env:d1_mini_pro]
> platform = espressif8266
> board = d1_mini
> framework = arduino
> build_flags = ${global.build_flags}
> extra_scripts = moveBinary.py
> upload_speed = 115200
> monitor_speed = 115200

So the assumption is that there is a dependency with 2.0.x and board definition of d1_mini_pro

thorsten-l, 2019-03-02:
i've the same issue.

bwint, 2019-03-03:
Same problem here using 2.0.0
board = d1_mini works but d1_mini_pro doesn't.

ivankravets, 2019-03-03:
Thanks, fixed! Please re-test with upstream version https://docs.platformio.org/en/latest/platforms/espressif8266.html#stable-and-upstream-versions

closed 2019-03-03 by ivankravets

jipp, 2019-03-03:
works with upstream version - thx
when will it be released?

ivankravets, 2019-03-03:
We are going to publish a new release within 24h. We work on https://github.com/platformio/platform-espressif8266/issues/134

philbowles, 2019-03-05:
Does this mean ALL incompatibilities between this and arduino board naming conventions are fixed? I really don't want to have to go through my code and change this:

```cpp
#if (defined(ARDUINO_ESP8266_WEMOS_D1MINI) || defined(ARDUINO_ESP8266_WEMOS_D1MINILITE) || defined(ARDUINO_ESP8266_WEMOS_D1MINIPRO))
```
etc etc for every ESP8266 board my codeeworks with...
..to add all the perversely different platformIO versions of the board names! Why are they not the same as the ARDUINO ones?

ivankravets, 2019-03-05:
@philbowles thanks that reported. We have just updated all macros in https://github.com/platformio/platform-espressif8266/commit/23d85b0d73d6faf10df2a76e75bb01acc671d553

Please update to 2.0.3 => https://github.com/platformio/platform-espressif8266/releases/tag/v2.0.3

commit 7e3d42a5 (2019-03-03): Fix broken LD script for "WeMos D1 mini Pro" board // Resolve #132