platformio/platform-espressif8266 #133: PIO doesn't flash init data to the right address for 1MB modules

maxgerhardt, 2019-03-01:
See https://community.platformio.org/t/checksum-errors-with-esp8266/6719/5 

Basically what's been observerd is that for 1MB ESP8266 modules, PlatformIO doesn't seem to flash `esp_init_data.bin` to the right address for the module (see https://nodemcu.readthedocs.io/en/master/en/flash/#sdk-init-data) causing a boot-failure. This was observed when using the non-RTOS SDK but might occur for other frameworks, too. 

The problem can be manually solved by downloading the needed init data from Espressif's SDK and flashing it. Please take a look at this to see if you can reproduce this and how to fix the problem.

maxgerhardt, 2019-03-05:
This seems to occur for *every* module type, not only for 1MB modules. The builder script simply doesn't flash this file?

ivankravets, 2019-03-06:
Please re-test with upstream version http://docs.platformio.org/en/latest/platforms/espressif8266.html#stable-and-upstream-versions

Does it work now?

maxgerhardt, 2019-03-06:
@ivankravets I've just erased the flash of my ESP12-E and reflashed it with the latest upstream version. The flashing command now includes `esp_init_data_default.bin` and the firmware works out-of-the box for NONOS framework.

```
esptool -cd nodemcu -cb 921600 -cp "COM9" -ca 0x00000 -cf C:\Users\Maxi\Desktop\wemos_test\.pioenvs\d1_mini_rtos\eagle.flash.bin -ca 0x10000 -cf C:\Users\Maxi\Desktop\wemos_test\.pioenvs\d1_mini_rtos\eagle.irom0text.bin -ca 0x3fc000 -cf C:\Users\Maxi\.platformio\packages\framework-esp8266-nonos-sdk\bin\esp_init_data_default.bin
```

However on ESP-RTOS-SDK, my blinky code for GPIO2 (built-LED on ESP12-E) doesn't work. The bootloader doesn't complain but my main code doesn't seem to be executed. So, there might be more to the issue. 

For Non-OS it's fixed however.

maxgerhardt, 2019-03-06:
@ivankravets also the code only has 2 possible addresses for the init data but there 6 module types? The code at 

https://github.com/platformio/platform-espressif8266/blob/ab839a2946998d0a916da12bee77e472a3959c61/builder/frameworks/esp8266-nonos-sdk.py#L104-L110 

won't cover all possibilities at https://nodemcu.readthedocs.io/en/master/en/flash/#sdk-init-data

maxgerhardt, 2019-03-06:
Ah, the missing key is maybe that we're not flashing `blank.bin` at the right address like mentioned in #62. Will test later in the day.

ivankravets, 2019-03-06:
Yes, I've just fixed all issues => https://github.com/platformio/platform-espressif8266/commit/ce5b99b4576129139a24144b740bc54ba47c7f8b

Does it work now? I tested on NodeMCUv2 and both RTOS and non-RTOS work for me. Can we have a bug fix release?

closed 2019-03-06 by ivankravets

maxgerhardt, 2019-03-06:
@ivankravets Works flawlessly for me now with NONOS and RTOS versions! My code was actually right and started executing after the latest fix commit. Tested with both my NodeMCUv2 and Wemos D1 mini.

ivankravets, 2019-03-06:
Thanks, released! https://github.com/platformio/platform-espressif8266/releases/tag/v2.0.4

commit ab839a29 (2019-03-06): Fix issue with broken examples for NONOS SDK and RTOS SDK // Resolve #62 Resolve #133
commit ce5b99b4 (2019-03-06): Flash blank.bin automatically for NONOS SDK and RTOS SDK // Resolve #62 Resolve #133