platformio/platform-espressif8266 #134: OTA port detection falsely triggers for serial ports

maxgerhardt, 2019-03-01:
It has been reported (see [here](https://community.platformio.org/t/upload-port-is-a-device-path-but-pio-tries-ota-upload/6781)) that PlatformIO attempts to use the OTA uploader for serial ports, such as `COM5`. 

This behaviour was made possible by this commit: https://github.com/platformio/platform-espressif8266/commit/54cd0fb1e2f61b1c6624d708f749372a37dd9648

Possible fixes:
* check against available serial ports, as written in that commit's message
* make it respect the `upload_protocol` option. The `main.py` script doesn't seem to do that? I'd expect it to use `esptool` when I say `upload_protocol = esptool` and not auto-detect the upload tool by resolving the upload port.

closed 2019-03-04 by ivankravets

ivankravets, 2019-03-04:
Thanks, Maximilian for the report. I've just rewritten upload logic in https://github.com/platformio/platform-espressif8266/commit/ae96496766b0aeeae60cfc81d45801b535bee66c and added support for `espota` protocol. This is the right solution.

tablatronix, 2020-05-19:
Why do I still have this problem ?
If I have no upload-flags set why is this not using the upload_tool from boards? 

nodemcuv2.upload.tool=esptool

```
[env:nodemcuv2_dev]
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = nodemcuv2
framework = arduino
build_flags = -DWM_DEBUG_PORT=Serial -DWM_DEBUG_LEVEL=3
lib_extra_dirs = /Users/shawn/projects/microcontrollers/dev/libraries
upload_speed = 921600
```

`pio run -t upload -e nodemcuv2_dev --upload-port /dev/cu.SLAB_USBtoUART`

```
Use manually specified: /dev/cu.SLAB_USBtoUART
"/Users/shawn/Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/bin/python" "/Users/shawn/.platformio/packages/tool-espotapy/espota.py" --debug --progress -i /dev/cu.SLAB_USBtoUART -f .pio/build/nodemcuv2_dev/firmware.bin
22:21:30 [DEBUG]: Options: {'timeout': 10, 'esp_ip': '/dev/cu.SLAB_USBtoUART', 'host_port': 14750, 'image': '.pio/build/nodemcuv2_dev/firmware.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
22:21:30 [INFO]: Starting on 0.0.0.0:14750
22:21:30 [INFO]: Upload size: 380720
```

Its really pissing me off

tablatronix, 2020-05-19:
ok so if it resolves it just changes it to espota, I am not setting upload_protocol in all my ini files, to hell with that, I guess ill rip this resolve crap out of main.py for now.

commit ae964967 (2019-03-04): Add "esptool" and "espota" to board configuration // Resolve #134