platformio/platform-espressif8266 #135: WiFi dropouts unless using WiFi.setSleepMode(WIFI_NONE_SLEEP)

CRCinAU, 2019-03-02:
I've noticed lately that multiple devices I have have started dropping off WiFi and reconnecting (causing MQTT dropouts etc) unless I specify `WiFi.setSleepMode(WIFI_NONE_SLEEP)` when setting up the WiFi adapter.

ESP.getFullVersion() output:
```
SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1/BearSSL:6778687
```

WiFi setup code in setup():
```
mConnectHandler = WiFi.onStationModeConnected(onConnected);
mDisConnectHandler = WiFi.onStationModeDisconnected(onDisconnect);
mGotIpHandler = WiFi.onStationModeGotIP(onGotIP);

WiFi.disconnect() ;
WiFi.persistent(false);
WiFi.setSleepMode(WIFI_NONE_SLEEP);
WiFi.mode(WIFI_STA);            // Client mode
WiFi.setOutputPower(17);        // 10dBm == 10mW, 14dBm = 25mW, 17dBm = 50mW, 20dBm = 100mW
WiFi.begin(ssid, password);     // Start WiFi.
```

And the functions:
```
void onConnected(const WiFiEventStationModeConnected& event){
update_status += millis();
update_status += ": Connected to AP.";
update_status += "\n";
}
void onDisconnect(const WiFiEventStationModeDisconnected& event){
update_status += millis();
update_status += ": Station disconnected";
update_status += "\n";
client.disconnect();
}
void onGotIP(const WiFiEventStationModeGotIP& event){
update_status += millis();
update_status += ": Station connected, IP: ";
update_status += WiFi.localIP().toString();
update_status += "\n";
timer.setTimeout(1000, checkForUpdate);
}

```
update_status is a global string that is included on the web page interface.

If I comment out the setSleepMode line, I get regular wifi disconnections.

pfeerick, 2019-04-18:
This is most likely related to the fact that the default for the wifi sleep type is MODEM_SLEEP,  a change in behaviour resulting from the ESP8266 Arduino core updating the underlying nonos-sdk and the introduction of the `setSleepMode` command in v2.5.0 which calls the nonos `wifi_set_sleep_type` function.  This is not PlatformIO related. If you have this issue with PlatformIO but not the Arduino IDE when using the same ESP8266 Arduino core version, that is a different kettle of fish entirely. So this issue should be closed, and instead raised over at https://github.com/esp8266/Arduino if something isn't working as expected. 

[image]

CRCinAU, 2019-04-18:
You are probably correct here. I've posted and linked the two issues. Closing this one.

closed 2019-04-18 by CRCinAU