platformio/platform-espressif8266 #136: OTA crashes device with Espressif8266 Platform >= 2.0.0

gabomdq, 2019-03-05:
Device: Lolin NodeMCU (ESP8266EX 4MB)

Doing the OTA procedure works fine with 1.8.0, but any platform release after that produces a crash on the device after the OTA connection is established and the first few bytes are uploaded (Device sometimes reports 2% upload, PlatformIO reports typically reports 1% progress before it halts).

The OTA code on device is the standard boilerplate ArduinoOTA stuff.

ManuelW77, 2019-04-29:
I can confirm, there is a memory heap. After a few mins the esp is running out of heap memory and crashes.

ivankravets, 2019-05-14:
Could you re-test with the latest upstream version http://docs.platformio.org/en/latest/platforms/espressif8266.html#stable-and-upstream-versions ?

closed 2019-09-02 by ivankravets