platformio/platform-espressif8266 #137: NONOS SDK platform not working after uploading to esp

iivancev, 2019-03-05:
Hi,

I am first time platformio user. Decided to check it out. It looked good, but after the problems I faced…i don’t think so anymore.

I am trying to upload blink example on nonos sdk platform for esp8266 (esp12). FIrst time it worked. After multiple (x100) attempts to reproduce sketch upload to work aagin, it fails every time.

I get inifinite amount of these:
ets Jan 8 2013,rst cause:2, boot mode:(3,6)

It is frustrating. Does anyone know what’s going on?

maxgerhardt, 2019-03-05:
Solved in https://community.platformio.org/t/using-plaformio-with-esp-nonos-sdk/6774 also refer to https://github.com/platformio/platform-espressif8266/issues/133

ivankravets, 2019-03-06:
Duplicate of https://github.com/platformio/platform-espressif8266/issues/133

closed 2019-03-06 by ivankravets