platformio/platform-espressif8266 #138: flash layout 1m64 for WEMOS D1 mini lite fails

thorsten-l, 2019-03-10:
In general i'm using 

platform = espressif8266
board = d1_mini_lite
framework = arduino
build_flags = -Wl,-Teagle.flash.1m64.ld

but nowadays this flash layout fails. I see watchdog resets via 115200 baud on the serial interface only.

---
ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
---

Without the "build_flags" everything works fine. But i need the custom flash layout.

ivankravets, 2019-05-14:
Does it work with Arduino IDE?

ivankravets, 2019-05-24:
Should be fixed in the upstream version. Please re-test with http://docs.platformio.org/en/latest/platforms/espressif8266.html#upstream

closed 2019-05-24 by ivankravets