platformio/platform-espressif8266 #141: Cannot build on android termux

ToRvaLDz, 2019-03-30:
What kind of issue is this?
- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Android arm7 termux

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.6.6

### Description of problem
Cannot build project

#### Steps to Reproduce

1. Install termux on android
2. run pkg install python-pip
2. Install platformio pip install platformio
3. Run platformio run in project folder

### Actual Results
```
$ platformio run
Processing test (platform: espressif8266@1.8.0; framework: arduino; board: esp01_1m)
---------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.1.3
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphome-core> 1.12.1
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.1.3
|   |   |-- <ESP8266WiFi> 1.0
Building .pioenvs/test/firmware.bin
Retrieving maximum program size .pioenvs/test/firmware.elf
Checking size .pioenvs/test/firmware.elf
sh: 1: esptool: not found
*** [.pioenvs/test/firmware.bin] Error 127
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  38.3% (used 31412 bytes from 81920 bytes)
PROGRAM: [===       ]  31.4% (used 321964 bytes from 1023984 bytes)
============================================================ [ERROR] Took 4.99 seconds ============================================================
$ platformio run -v
Processing test (platform: espressif8266@1.8.0; build_flags: -DESPHOME_USE, -DUSE_ADC_SENSOR, -DUSE_API, -DUSE_BINARY_SENSOR, -DUSE_DALLAS_SENSOR, -DUSE_DHT_SENSOR, -DUSE_GPIO_BINARY_SENSOR, -DUSE_GPIO_SWITCH, -DUSE_NEW_OTA, -DUSE_OTA, -DUSE_SENSOR, -DUSE_STATUS_BINARY_SENSOR, -DUSE_STATUS_LED, -DUSE_SWITCH, -DUSE_TEMPLATE_BINARY_SENSOR, -DUSE_TEMPLATE_SENSOR, -DUSE_TEMPLATE_SWITCH, -DUSE_WIFI_SIGNAL_SENSOR, -Wl,-Teagle.flash.1m0.ld, -Wno-sign-compare, -Wno-unused-but-set-variable, -Wno-unused-variable; lib_deps: ESP8266WiFi, ESP8266mDNS, ESPAsyncTCP@1.1.3, Hash, https://github.com/esphome/esphome-core/archive/v1.12.1.zip; lib_ignore: ArduinoOTA, Update, Wire, FastLED, NeoPixelBus, ESP Async WebServer, AsyncMqttClient, AsyncTCP; framework: arduino; upload_speed: 115200; board: esp01_1m; board_build.flash_mode: dout; lib_ldf_mode: chain)
---------------------------------------------------------------------------------------------------------------------------------------------------
Library Storage: /data/data/com.termux/files/home/test/.piolibdeps
Looking for ESP8266WiFi library in registry
Warning! Library `{'requirements': None, 'name': 'ESP8266WiFi'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'ESP8266WiFi'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Looking for ESP8266mDNS library in registry
Warning! Library `{'requirements': None, 'name': 'ESP8266mDNS'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'ESP8266mDNS'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
LibraryManager: Installing id=305 @ 1.1.3
ESPAsyncTCP @ 1.1.3 is already installed
[... 122 lines cut ...]
  WiFiComponent *wificomponent = App.init_wifi();
  wificomponent->set_use_address("test.local");
  WiFiAP wifiap = WiFiAP();
  wifiap.set_ssid("xxxx");
  wifiap.set_password("xxxxxx");
  wificomponent->add_sta(wifiap);
  OTAComponent *otacomponent = App.init_ota();
  otacomponent->set_auth_password("test");
  otacomponent->start_safe_mode();
  api::APIServer *api_apiserver = App.init_api_server();
  api_apiserver->set_password("test");
  // =========== AUTO GENERATED CODE END ============
  // ========= YOU CAN EDIT AFTER THIS LINE =========
  App.setup();
}

void loop() {
  App.loop();
}
```

### Additional info
The project is generated from esphome.

Thanks

ivankravets, 2019-03-30:
Can you try 0.4.12 version of esptool? https://github.com/igrr/esptool-ck/releases

Please replace file `~/.platformio/packages/tool-esptool/esptool` by https://github.com/igrr/esptool-ck/releases/download/0.4.12/esptool-0.4.12-linux-armhf.tar.gz

Does it work?

ToRvaLDz, 2019-03-30:
No, it doesn't work, probably something related to chroot

ToRvaLDz, 2019-03-30:
Building it from https://github.com/igrr/esptool-ck/ solve the issue

Thank you for driving me in the right direction.

closed 2019-03-30 by ToRvaLDz

CODeRUS, 2020-11-17:
Does it work for anyone on Android Termux nowadays? My setup is missing all the xtensa tools, compilers, etc.

yesco, 2021-09-20:
No, it doesn't work to try to build any esp32 code using platform.io under Termux.

I get, it's nice and retro with numeric error messages and undocumented negative error codes... 

xtensa-esp32-elf-gcc .....
 *** [.pio/build/esp32dev/libFrameworkArduinoVariant.a] Error -31

I don't know but last I tried to look at it it seems the xtensa-gcc or linker is dumping core! Haha

DMaisterra, 2022-03-29:
> Building it from https://github.com/igrr/esptool-ck/ solve the issue
> 
> Thank you for driving me in the right direction.

And how you compile, its good to ulpoad code in the go, thats also can be made with bluino loader.
I think the the cool stuff its also compile so.

export PATH=$PATH:$HOME/.platformio/penv/bin
that peace of line solves the errors that yesco's out.
but in the time you added to .bashrc. its again the later.... in diferents ways ex diferents errors.
All of them things "pio" dont found.