platformio/platform-espressif8266 #143: Need to add .text1 segment to generated binary

earlephilhower, 2019-04-11:
The linker in the 8266 Arduino code needed to split the .TEXT segment into two separate segments to support moving some std::function methods into IRAM (as they are used in IRQ callbacks for certain interfaces).  See https://github.com/esp8266/Arduino/pull/5922

There are now ".text" and ".text1" segments which comprise the program flash area.  Our ELF2BIN.PY script has been updated to support this transparently, but it looks like PIO uses its own method of generating BIN files.  Without the .text1 segment in the BIN image, the app obviously won't run and you'll get a boot loop.  See https://github.com/esp8266/Arduino/issues/5974

My quick scan makes me think the following two spots need to have `.text1` added after `.text` to generate a proper image.

https://github.com/platformio/platform-espressif8266/blob/3e7759744d13d9bb60f16f35a6b2e49d049f5939/builder/main.py#L229

https://github.com/platformio/platform-espressif8266/blob/3e7759744d13d9bb60f16f35a6b2e49d049f5939/builder/main.py#L250

Thx
-EFP3

Jason2866, 2019-04-11:
With this changes, Tasmota builds are working again.
Thx EFP3!

earlephilhower, 2019-04-12:
Also need to add `.text1` to the size computation, or it won't capture the full program usage in output:
https://github.com/platformio/platform-espressif8266/blob/3e7759744d13d9bb60f16f35a6b2e49d049f5939/builder/main.py#L151

PR #144 (2019-04-14): Include TEXT1 in generated bins, avoid boot loop
The Arduino 8266 code recently split .text into .text and .text1 to
enable placing certain functions in IRAM automatically (required for
callbacks in IRQs).

Update the binary creation and size calculation to account for this.

Without this change, infinite boot loops will occur because the main
code will never be uploaded to the chip.

Fixes #143
commit 23ea21c5 (2019-04-14): Include TEXT1 in generated bins, avoid boot loop (#144)

The Arduino 8266 code recently split .text into .text and .text1 to
enable placing certain functions in IRAM automatically (required for
callbacks in IRQs).

Update the binary creation and size calculation to account for this.

Without this change, infinite boot loops will occur because the main
code will never be uploaded to the chip.

Fixes #143
closed 2019-04-14 by ivankravets