platformio/platform-espressif8266 #146: ESP8266: PlatformIO compiles not like ArduinoIDE (guess related to #83)

pauldec, 2019-04-28:
I'm just starting to use PlatformIO on VSC and I'm fairly new to programming, Uploading my code with the Arduino IDE gives the correct result in the webbrowser, unfortunately uploading the same code with PlatformIO doesn't give the correct result. The code starts (I get the correct messages in the serial monitor) but stops after startup (no webpage is displayed). Adding the build_flags = -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY to platformio.ini changed the problem in a positive way, now the webpage is diplayed sometimes but always stops reacting after one or two clicks in the web interface. 
Version: 1.33.1 (user setup)
Commit: 51b0b28134d51361cf996d2f0a1c698247aeabd8
Date: 2019-04-11T08:27:14.102Z
Electron: 3.1.6
Chrome: 66.0.3359.181
Node.js: 10.2.0
V8: 6.6.346.32
OS: Windows_NT x64 10.0.17134

PlatformIO v. 4.0.0a8

[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
monitor_port = COM4
monitor_speed = 115200
upload_speed = 115200
upload_protocol = esptool   
build_flags = -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
[esp8266_compile.txt](https://github.com/platformio/platform-espressif8266/files/3125476/esp8266_compile.txt)
Thanks for any advise.

ivankravets, 2019-05-14:
What is your configuration in Arduino IDE?

pauldec, 2019-05-15:
Thank you for your help on this,
Currently I'm using Arduino IDE 1.8.9 (App 1.8.21.0). I did a clean install (uninstalled Arduino, deleted Arduino15 folder, rebooted). 
The good news: I found that adding a delay(100) after server.handleClient() solves the issue (in the original Arduino IDE code there is no delay()) , is this possible?
Regards, Paul.

ivankravets, 2019-05-15:
Do you use the same board configuration in Arduino IDE? Please compare with http://docs.platformio.org/en/latest/platforms/espressif8266.html#configuration

pauldec, 2019-05-15:
I think so, this is what PlatformIO displays while uploading:
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
In Arduino IDE: 80Mhz; 4M/1M; lower memory
see also attached image
[image]
platformio.ini
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
monitor_speed = 115200

ivankravets, 2019-05-24:
Please re-test with the latest 2.2.1 release. Does it work now?

pauldec, 2019-05-25:
Thank you, it does work now even without the delay() after server.handleClient().

pauldec, 2019-05-25:
In my real code (where I use an attachInterrupt) I now receive an "ISR not in IRAM" error. Adding the interrupt in the testcode also gives this kind of error.
Found this link about ISR not in IRAM: (https://www.esp8266.com/viewtopic.php?p=82454).
Tested the following: v2.0.0, 2.0.1 and 2.0.4 seem to need the delay() this delay() may depend upon the lines of code after server.handleClient(), v2.1.0 and above seem to give the ISR error. 
Because I wanted to send mails with Gsender _(2019 Update: This code doesn't work with ESP8266 core for Arduino version 2.5.0! Temporary solution use core version 2.4.2)_ , I had to downgrade to v1.8.0.
And, happy me, v1.8.0 seems to be the solution for my code.
`[env:d1_mini]
; platform = espressif8266
platform = espressif8266@1.8.0
; platform = espressif8266@2.0.0
; platform = espressif8266@2.2.1
; platform = espressif8266@2.2.0
; platform = espressif8266@2.1.0
; platform = espressif8266@2.0.4
; platform = espressif8266@2.0.1
board = d1_mini
framework = arduino
monitor_port = COM4
monitor_speed = 115200`
Thank you very much for your help, I now understand a tiny little bit of what is going on behind the compile/upload process.
Regards, paul.

closed 2019-05-25 by pauldec