platformio/platform-espressif8266 #147: Feature Request: Arduino bugfix v.2.5.1 is released

Jason2866, 2019-05-11:
Since v.2.5.1 is released could you please update to this version?
Thx!

ivankravets, 2019-05-13:
@valeros could you investigate why does not work the latest source code v2.5.1 of Arduino core? it builds OK but does not work after flashing.

@d-a-v, @earlephilhower do we have critical changes in 2.5.1 build process? Do we need to upgrade to the latest ESP8266 toolchain and switch off from esptool-ck to esptool-py? Thanks!

earlephilhower, 2019-05-13:
@ivankravets There was a linking change (.text into .text and .text1 segment to allow moving some specific functions to IRAM as needed for IRQ callbacks) but I think we already got you in the loop on it.

https://github.com/esp8266/Arduino/pull/5922

Jason2866, 2019-05-13:
@ivankravets  @valeros 
I get a working build with old and new xtensa build chain. No problem with arduino core.
You can take a look here:
`platform                  = https://github.com/Jason2866/platform-espressif8266.git#Stage-Tasmota`
for new extensa build chain this file has to be adapted (linked files has to be reordered)
https://github.com/Jason2866/TDM/blob/master/platformio.py
and old Xtensa build chain
`platform                  = https://github.com/Jason2866/platform-espressif8266.git#Tasmota`

mcspr, 2019-05-13:
Just to make clear what "linked files has to be reordered" means, see
Core 2.5.0 / platform 2.0.0 issue: https://github.com/platformio/platform-espressif8266/issues/128#issuecomment-466615033 (and further comments)
workaround for that: https://github.com/mcspr/platformio-core/commit/708eff208877cefbe7d05805bc65ab975f723e3b

ivankravets, 2019-05-13:
@mcspr why do you need this hook https://github.com/mcspr/platformio-core/commit/708eff208877cefbe7d05805bc65ab975f723e3b ?

Jason2866, 2019-05-13:
Without this hook code doesnt boot builded with new xtensa build chain.
@mcspr thx for you fix, searched the orig. issue and didnt found

mcspr, 2019-05-13:
Yes, like @Jason2866 said - updated toolchain seemingly treats -Wl,-T and just -T differently.
[esp8266/Arduino never encountered that, because platform.txt / arduino-builder only uses -T{user_selected_ldscript}](https://github.com/esp8266/Arduino/blob/98125f88605cd7e46e9be4e1b3ad0600dd5d2b51/platform.txt#L54)

ivankravets, 2019-05-13:
@Jason2866 @mcspr this is a bug with ESP8266 toolchain. Please file an issue here https://github.com/earlephilhower/esp-quick-toolchain and I'm sure @earlephilhower will take a look.

See GCC docs https://gcc.gnu.org/onlinedocs/gcc/Link-Options.html

> -Wl,option
Pass option as an option to the linker. If option contains commas, it is split into multiple options at the commas. You can use this syntax to pass an argument to the option. For example, -Wl,-Map,output.map passes -Map output.map to the linker. When using the GNU linker, you can also get the same effect with -Wl,-Map=output.map.

PlatformIO Core is used for over 30 dev/platform which depend on the different toolchains. It would be great if we can keep PlatformIO Core compatible with GCC CLI.

earlephilhower, 2019-05-13:
Guys, the toolchain got bumped in 2.5.0 and was not touched significantly for 2.5.1 AFAIR.  Was 2.5.0 working?

If so, then I really think it's something else because we're building the unadulterated GNU sources for the vast majority of things (minor patches, unrelated to the linker).

Jason2866, 2019-05-13:
@earlephilhower  the toolchain from 2.5.0 is working fine for 2.5.1 with this change from @mcspr

ivankravets, 2019-05-13:
@earlephilhower the bug was introduced in this release https://github.com/earlephilhower/esp-quick-toolchain/releases/tag/2.5.0-3

We use a toolchain that was before "2.5.0-3". The latest "2.5.0-3" breaks ESP8266 SDKs which we support in PlatformIO.

earlephilhower, 2019-05-13:
@ivankravets what is the exact version you have been using?  Again, we are building the plain GCC code here so there's not much I expect to be able to track down.  Patches are kept in a separate dir in the repo, and none relate to the g++/ld options parsing...

Jason2866, 2019-05-13:
@ivankravets  anyway, with the old build chain (which is standard for Arduino ESP8266) the code compiles fine and does work.
`platform = https://github.com/Jason2866/platform-espressif8266.git#Tasmota`

earlephilhower, 2019-05-13:
Also, does PIO use its own bootloader on-chip?  Like I said, there is another segment in the generated binary now (iram.text1) which needs to be supported.  The Arduino bootloader handles it seemlessly (w/o any changes), but if you've got a custom one it needs to be aware of this and support it.

octavio2895, 2019-05-13:
Just to add my few cents. I'm having some issues with OTA updates using the stable release. It appears that espota thinks its done sending the sketch before the fact and I get an Error[4]: End Failed on the serial output. The sketch gets to about 89% done uploading before it fails. I tried using the 2.5.0 release of the core in the arduino IDE and it works great, so I don't think the core is the problem. After changing the platform using https://github.com/platformio/platform-espressif8266/issues/147#issuecomment-491932097 the uploads completes most of the times. However, when I switched to an AP with a stronger signal, the fail rate drops with stable and with https://github.com/platformio/platform-espressif8266/issues/147#issuecomment-491932097 .

Does OTA update work differently in platformio from arduino ide?

I wasn't sure if I should open a new issue, so apologies if I should have.

closed 2019-05-13 by ivankravets

ivankravets, 2019-05-13:
@octavio2895 I've just switch espota to official script from Arduino core. Please re-test with upstream version http://docs.platformio.org/en/latest/platforms/espressif8266.html#upstream

ivankravets, 2019-05-13:
@earlephilhower @Jason2866 @mcspr

I've just switch ElfToBin builder and upload process to esptool.py:
- https://github.com/esp8266/Arduino/pull/6091
- https://github.com/platformio/platform-espressif8266/commit/23542cf37e2410ed59fd27bdc5fb7d7d278fc2f8

It works for me now. Please re-test with upstream version of dev/platform http://docs.platformio.org/en/latest/platforms/espressif8266.html#upstream

if it works for you too, we will make a final release.

Thanks!

earlephilhower, 2019-05-13:
That commit uses the new elf2bin.py script which knows about the add'l .text1 section.  If that really works, then the -Wl was a red herring and we can ignore it in the core.

octavio2895, 2019-05-13:
@ivankravets that fixed the issue! where can I find the version of espota.py that was being used before? I would like to take a look at it.

Also, should I make an issue for this? since its not fixed on the stable release to help people with similar problems?

mcspr, 2019-05-14:
Can confirm that elf2bin change is working with platform from `develop` branch.

platform.json:["packages"]["toolchain-xtensa"]["version"] is still pointing to the ~1.40802.0, so toolchain problem obviously is not there.
Original issue is still reproducible with the ~2.40802.0 (toolchain package version comes from [platfromio's package repo](https://dl.bintray.com/platformio/dl-packages/manifest.json). it is the same as from [esp-quick-toolchain 2.5.0-3 release](https://github.com/earlephilhower/esp-quick-toolchain/releases/download/2.5.0-3/x86_64-linux-gnu.xtensa-lx106-elf-20ed2b9c.tar.gz), same build-id shas and everything)

Jason2866, 2019-05-14:
Can confirm develop is working. Feature request with this issue is done. Thx.

Issue with toolchain ~2.4 is still there (when used) as @mcspr  mentioned.

The develop version uses ~1.4 build chain:
```
PlatformManager: Installing platform-espressif8266
git version 2.19.1.windows.1
Cloning into 'C:\Users\Hans\.platformio\platforms\_tmp_installing-jnoeh3-package'...
remote: Enumerating objects: 95, done.
remote: Counting objects: 100% (95/95), done.
remote: Compressing objects: 100% (74/74), done.
remote: Total 95 (delta 50), reused 34 (delta 10), pack-reused 0
Unpacking objects: 100% (95/95), done.
espressif8266 @ e7d8b22 has been successfully installed!
The platform 'https://github.com/platformio/platform-espressif8266.git#develop' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing toolchain-xtensa @ ~1.40802.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
PackageManager: Installing tool-esptool @ <2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
PackageManager: Installing tool-esptoolpy @ ~1.20600.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
PackageManager: Installing framework-arduinoespressif8266 @ ~2.20501.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100% 

```

ivankravets, 2019-05-15:
Please update to the stable version of dev/platform https://github.com/platformio/platform-espressif8266/releases/tag/v2.1.0

Jason2866, 2019-05-15:
Release does install and compile results in a working firmware of tried project [Tasmota](https://github.com/arendst/Sonoff-Tasmota)

commit ec1a8a1a (2019-05-13): Update Arduino core to 2.5.1 // Resolve #147