platformio/platform-espressif8266 #148: Problem in FS after updating the core

lucasromeiro, 2019-05-15:
Hello, I'm using PlatformIO in its latest version. 
I have now updated to use the Arduino 2.5.1 core to use ESP8266 12-F and now my code does not compile anymore!
Accuses error in FS.
I do not understand why. It worked before updating.
I can not find the problem.
Can someone help me?

[image]
[image]

Thank you

lucasromeiro, 2019-05-15:
core 2.4.2 work fine:
platform = espressif8266@1.8.0

core 2.5.1 don't work:
platform = espressif8266

why??

ivankravets, 2019-05-15:
> Unification of SPIFFS and SD, into single compatible FS, File, and Dir objects

See https://github.com/esp8266/Arduino/pull/5525

@earlephilhower do you have any migrating guide?

earlephilhower, 2019-05-15:
There is no migration.  They are compatible AFAIK and per many other users and the original SD.h examples, even.

ivankravets, 2019-05-15:
@lucasromeiro could you share a simple project to reproduce this issue?

lucasromeiro, 2019-05-15:
> @lucasromeiro could you share a simple project to reproduce this issue?

Sure!

#include "Arduino.h"
#include <ESP8266WiFi.h>
#include <FS.h>          //Biblioteca responsável por criar o Sistema de manipulação de arquivos

void setup() {
  SPIFFS.begin(); //Inicializa sistema de arquivos
}

void loop() {
  
  File carregaArquivos = SPIFFS.open("/log.txt", "a");
  char charState = '1';
  carregaArquivos.write(charState);
  carregaArquivos.close();
}

[image]

earlephilhower, 2019-05-15:
fs::write takes a pointer and length, or a c-string, or a String, not a char.

lucasromeiro, 2019-05-15:
> fs::write takes a pointer and length, or a c-string, or a String, not a char.

If I use String it also does not work!
String charState = "1";

lucasromeiro, 2019-05-15:
I do not understand why this worked in the previous version and now it does not work.

earlephilhower, 2019-05-15:
Mybad, that is not part of the function signatures.

::write is different than ::print.  Maybe you meant to use ::print instead?

earlephilhower, 2019-05-15:
Actually, I think I see the issue.

FS::File never had a File::write(char) method, but it did have a File::write(uint8_t) method.  Print(the superclass of File) never supported Print::write(char).

So in the old versions you were getting silent conversions of char->uint8_t (I think you'll get a warning if you go strict enough).

Now, since your call doesn't match any exactly, it is using the template which is for writing Stream children, only.

earlephilhower, 2019-05-15:
Try the change in https://github.com/esp8266/Arduino/pull/6101 .  I'm not sure if it will break anything else, but it re-enabled the usage.

lucasromeiro, 2019-05-15:
> Actually, I think I see the issue.
>
> FS::File never had a File::write(char) method, but it did have a File::write(uint8_t) method. Print(the superclass of File) never supported Print::write(char). [...]

Got it, so if I switch char to uint8_t it should work the same, right?
I'll try.

lucasromeiro, 2019-05-16:
> Actually, I think I see the issue.
>
> FS::File never had a File::write(char) method, but it did have a File::write(uint8_t) method. Print(the superclass of File) never supported Print::write(char). [...]

Its work!

uint8_t charState = '1';

Thanks!

closed 2019-05-16 by lucasromeiro