platformio/platform-espressif8266 #149: After upgrade: esptool write_flash error:

MacWyznawca, 2019-05-15:
After upgrade I can't write flash to ESP8266 (nonos).

PROGRAM: [=======   ]  70.2% (used 264389 bytes from 376832 bytes)
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /dev/cu.usbserial-AH03B6IO
Uploading .pioenvs/d1_mini/eagle.flash.bin
usage: esptool write_flash [-h] [--erase-all]
[--flash_freq {keep,40m,26m,20m,80m}]
[--flash_mode {keep,qio,qout,dio,dout}]
[--flash_size FLASH_SIZE]
[--spi-connection SPI_CONNECTION] [--no-progress]
[--verify] [--compress | --no-compress]
<address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x0 for file: /Users/jkk/Documents/HomeKit/ESP-NOW/ESP-NOW Termostat/.pioenvs/d1_mini/eagle.flash.bin
*** [upload] Error 2

WaldoPepper, 2019-05-16:
...same here, toolchain is:

espressif8266 @ 2.1.0
toolchain-xtensa @ 1.40802.0
tool-esptool @ 1.413.0
tool-esptoolpy @ 1.20600.0

MacWyznawca, 2019-05-16:
I can flash compiled program using command:

`esptool.py` -p /dev/cu.usbserial-AH03B6IO write_flash 0x00 eagle.flash.bin -fm dio -ff 40m

`esptool.py` -p /dev/cu.usbserial-AH03B6IO write_flash 0x10000 eagle.irom0text.bin -fm dio -ff 40m

and it works ok. Only Platformio can't write firmware.

`/dev/cu.usbserial-AH03B6IO` is my USB-UART port on macOS

ivankravets, 2019-05-16:
Sorry for the issue. Please update to 2.1.1 => https://github.com/platformio/platform-espressif8266/releases/tag/v2.1.1

commit 0d2518f5 (2019-05-16): Fix "esptool write_flash error" when using ESP8266 SDK // Resolve #149
closed 2019-05-16 by ivankravets