platformio/platform-espressif8266 #150: Floating point exception (core dumped)

Riyuzakii, 2019-05-15:
I'm trying to upload firmware to Odroid XU4, SmartPower 2.
I made a change to increase logging frequency of power consumption data.
The code for firmware was taken from here: [https://github.com/hardkernel/smartpower2.git](https://github.com/hardkernel/smartpower2.git)
These two commands run fine(with some warnings about some enumeration value and unused variables):
```
sudo platformio run
sudo platformio run --target upload
```
when I run the following command, I'm getting the shown error:

```
sudo platformio run --target uploadfs

Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 160MHz 80KB RAM (4MB Flash)
Converting smartpower2.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WebSockets> 2.1.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <SPI> 1.0
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0
|-- <LiquidCrystal_I2C>
|   |-- <Wire> 1.0
|-- <mcp4652>
|   |-- <Wire> 1.0
|-- <Hash> 1.0
|-- <ESP8266WiFi> 1.0
|-- <SimpleTimer>
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
Building SPIFFS image from 'data' directory to .pioenvs/esp12e/spiffs.bin
Floating point exception (core dumped)
*** [.pioenvs/esp12e/spiffs.bin] Error 136
====================================================================== [ERROR] Took 0.76 seconds ======================================================================

```
Can someone help me out with this? There's another [issue](https://github.com/platformio/platformio-core/issues/469) like this but that issue doesn't really address the problem plus the error code is different.
Thanks

ivankravets, 2019-05-16:
1. Please don't use `sudo` for PlatformIO. I highly recommend to remove `~/.platformio` folder and start using PlatformIO Core without sudo. It's dangerous.
2. Please run the same command but in verbose mode:
```
platformio run --target uploadfs -v
```

Riyuzakii, 2019-05-16:
ran without sudo, 
`platformio run --target uploadfs -v

Processing esp12e (board_build.f_cpu: 160000000L; build_flags: -Wl,-Tesp8266.flash.4m.ld; lib_deps: WebSockets, Simpletimer, LiquidCrystal_I2C; platform: espressif8266; upload_speed: 921600; board: esp12e; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Storage: /home/light/git/smartpower2-master/.piolibdeps
LibraryManager: Installing id=549
WebSockets @ 2.1.3 is already installed
Looking for Simpletimer library in registry
Found: https://platformio.org/lib/show/419/SimpleTimer
LibraryManager: Installing id=419
SimpleTimer @ b30890b8f7 is already installed
LibraryManager: Installing id=576
LiquidCrystal_I2C @ 1.1.4 is already installed
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 160MHz 80KB RAM (4MB Flash)
xtensa-lx106-elf-g++ -o "/home/light/git/smartpower2-master/src/smartpower2.ino.cpp" -x c++ -fpreprocessed -dD -E "/tmp/tmpyK1RaI"
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WebSockets> 2.1.3 (/home/light/git/smartpower2-master/lib/WebSockets_ID549)
|   |-- <ESP8266WiFi> 1.0 (/home/light/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi)
|   |-- <Hash> 1.0 (/home/light/.platformio/packages/framework-arduinoespressif8266/libraries/Hash)
|   |-- <Ethernet(esp8266)> 1.0.4 (/home/light/.platformio/packages/framework-arduinoespressif8266/libraries/Ethernet)
|   |   |-- <SPI> 1.0 (/home/light/.platformio/packages/framework-arduinoespressif8266/libraries/SPI)
|   |-- <SPI> 1.0 (/home/light/.platformio/packages/framework-arduinoespressif8266/libraries/SPI)
|-- <LiquidCrystal_I2C> (/home/light/git/smartpower2-master/lib/LiquidCrystal_I2C_ID576)
|   |-- <Wire> 1.0 (/home/light/.platformio/packages/framework-arduinoespressif8266/libraries/Wire)
|-- <Wire> 1.0 (/home/light/.platformio/packages/framework-arduinoespressif8266/libraries/Wire)
|-- <ESP8266WiFi> 1.0 (/home/light/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi)
|-- <Hash> 1.0 (/home/light/.platformio/packages/framework-arduinoespressif8266/libraries/Hash)
|-- <mcp4652> (/home/light/git/smartpower2-master/lib/mcp4652)
|   |-- <Wire> 1.0 (/home/light/.platformio/packages/framework-arduinoespressif8266/libraries/Wire)
|-- <SimpleTimer> (/home/light/git/smartpower2-master/lib/SimpleTimer_ID419)
|-- <ESP8266WebServer> 1.0 (/home/light/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WebServer)
|   |-- <ESP8266WiFi> 1.0 (/home/light/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi)
"mkspiffs" -c data -p 0 -b 0 -s 0 .pioenvs/esp12e/spiffs.bin
Floating point exception (core dumped)
*** [.pioenvs/esp12e/spiffs.bin] Error 136
====================================================================== [ERROR] Took 0.75 seconds ======================================================================
`

ivankravets, 2019-05-16:
> -Wl,-Tesp8266.flash.4m.ld

You are going to use LD script without SPIFFS space. Please read:

- http://docs.platformio.org/en/latest/platforms/espressif8266.html#uploading-files-to-file-system-spiffs
- http://docs.platformio.org/en/latest/platforms/espressif8266.html#platform-espressif-customflash
- https://github.com/esp8266/Arduino/tree/master/tools/sdk/ld

You can try to replace `-Wl,-Tesp8266.flash.4m.ld` with `-Wl,-Teagle.flash.4m1m.ld`

closed 2019-05-16 by ivankravets

Riyuzakii, 2019-05-16:
thanks, @ivankravets