platformio/platform-espressif8266 #153: 2.1.1 - Error 2 on flashing the firmware (Wemos D1)

owah, 2019-05-21:
Hello,

I have an issue similar to https://github.com/platformio/platform-espressif8266/issues/149, namely that I can't upload firmware to a Wemos D1 mini anymore, after upgrading.

Checking size .pioenvs/d1_mini/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  35.9% (used 29436 bytes from 81920 bytes)
PROGRAM: [====      ]  38.4% (used 401576 bytes from 1044464 bytes)
Creating BIN file ".pioenvs/d1_mini/firmware.bin" using ".pioenvs/d1_mini/firmware.elf"
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.usbserial-1460
Uploading .pioenvs/d1_mini/firmware.bin
esptool.py v2.6
Serial port /dev/cu.usbserial-1460
Connecting....
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...

A fatal error occurred: Invalid head of packet (0xE0)
*** [upload] Error 2

If I manually change the `upload_speed` in `platformio.ini` to: `upload_speed = 115200` then I get a successful upload:

Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 405728 bytes to 294493...
Wrote 405728 bytes (294493 compressed) at 0x00000000 in 28.6 seconds (effective 113.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

My toolchain is the current toolchain (2.1.1)

owah, 2019-05-21:
So I just noticed there was an update released a few minutes ago. I have updated my toolchain:

Updating espressif8266                   @ 2.2.0          [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20502.0      [Up-to-date]
Updating toolchain-xtensa                @ 2.40802.190218 [Up-to-date]

But the firmware uploading issue persists

closed 2019-05-24 by ivankravets

pfeerick, 2019-07-28:
Can I ask why 115200 was picked? As noted in both the[ arduino-esp8266 PR](https://github.com/esp8266/Arduino/pull/5635#issuecomment-495034459) changing to esptool.py, and the [esptool repo](https://github.com/espressif/esptool/issues/435), 460800 seemed to be a reliable default, until they can work out why 921600 doesn't work (but 100000 does).

Orel-A, 2020-12-29:
115200 is too slow! minimum should be 460800 as in line with Arduino library. esp-tool is already in version 3.0

commit 93ac4f9a (2019-05-24): Use 115200 upload speed by default // Resolve #153