platformio/platform-espressif8266 #160: Interruption problem with esp8266 (nodemcu; nodemcuv2, d1_mini . I have checked for them)

alexgalstyan, 2019-06-14:
Hello Ivan,
I have found problem with interruption in PlatformIO compiling: 

platformio settings:

`[env:nodemcu]
platform = espressif8266
board = nodemcu
framework = arduino
; COM-порт для загрузки
upload_port = COM3
; Скорость загрузки в baud
upload_speed = 115200 ;921600 ;
monitor_speed = 115200

`
`
//small sketch for checkin
#include <Arduino.h>

const byte interruptPin = D5; //can use any PIN for test, the problem persists for all
volatile byte interruptCounter = 0;
int numberOfInterrupts = 0;
 
void handleInterrupt() {
  interruptCounter++;
}

void setup() {
 
  Serial.begin(115200);
  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin), handleInterrupt, FALLING);
 
}
 
void loop() {
 
  if(interruptCounter>0){
 
      interruptCounter--;
      numberOfInterrupts++;
 
      Serial.print("An interrupt has occurred. Total: ");
      Serial.println(numberOfInterrupts);
  }
 
}`

result by serial getting:
Processing nodemcu (framework: arduino; platform: espressif8266; board: nodemcu)
---------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcu.html
PLATFORM: Espressif 8266 > NodeMCU 0.9 (ESP-12 Module)
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 34 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs\nodemcu\src\main.cpp.o
Retrieving maximum program size .pioenvs\nodemcu\firmware.elf
Checking size .pioenvs\nodemcu\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  32.9% (used 26916 bytes from 81920 bytes)
PROGRAM: [===       ]  25.2% (used 263392 bytes from 1044464 bytes)
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: COM3
Uploading .pioenvs\nodemcu\firmware.bin
esptool.py v2.6
Serial port COM3
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: b4:e6:2d:45:21:ce
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 267552 bytes to 195036...
Wrote 267552 bytes (195036 compressed) at 0x00000000 in 17.3 seconds (effective 124.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
========================================================== [SUCCESS] Took 22.61 seconds ==========================================================
--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
rd␀$��|␀�l�<␃␌␄␌�␄l�␌#|��␃�␛�s�c�␌b��og�l'o���␌b␜x��lrl;l8�g�␐␃␄␄�␄l␄��␄␌␄c␄o�|␃l�␌␄�c��no�␀$��l`␃�␓␛g'␌d`␃␇␃gs�ۓo␌␄b␄�␏$␏r��'␄␌c␌�␇d�␃�␃ld�␓�l`␃��o�␃ISR not in IRAM!

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffdb0 end: 3fffffc0 offset: 01b0
3fffff60:  feefeffe 00000001 3ffee26c 3ffee2c4
3fffff70:  3fffdad0 00000001 3ffee26c 3ffee2c4
3fffff80:  3fffdad0 00000000 3ffee294 401003de
3fffff90:  feefeffe 00000000 3ffee294 40201069
3fffffa0:  feefeffe feefeffe feefeffe 402018c0
3fffffb0:  feefeffe feefeffe 3ffe84f4 401006b1
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
ISR not in IRAM!

alexgalstyan, 2019-06-14:
I have found some solve in GitHub forums 

[1.] (https://github.com/sandeepmistry/arduino-LoRa/issues/254)
[2.](https://github.com/LowPowerLab/RFM69/issues/124)
[3.](https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L661-L685)

need to add ICACHE_RAM_ATTR for definitions ESP822 or ESP32
, or need use old core  for example 2.0.4

alexgalstyan, 2019-06-14:
Previous example with working condition:
`#include <Arduino.h>

// #ifdef BOARD_ESP8266
//     #define ISR_PREFIX ICACHE_RAM_ATTR 
// #elif BOARD_SOMEOTHER
//     #define ISR_PREFIX FAST_MEM_LOCATION
// #else 
//     #define ISR_PREFIX
// #endif 

#ifdef ESP8266 || ESP32
    #define ISR_PREFIX ICACHE_RAM_ATTR
#else
    #define ISR_PREFIX
#endif

const byte interruptPin = D5;
volatile byte interruptCounter = 0;
int numberOfInterrupts = 0;
 
ISR_PREFIX void handleInterrupt() {
  interruptCounter++;
}

void setup() {
 
  Serial.begin(115200);
  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin), handleInterrupt, FALLING);
 
}
 
 
void loop() {
 
  if(interruptCounter>0){
 
      interruptCounter--;
      numberOfInterrupts++;
 
      Serial.print("An interrupt has occurred. Total: ");
      Serial.println(numberOfInterrupts);
  }
 
}`

closed 2019-06-14 by alexgalstyan

sabaEbr, 2020-07-13:
Thank very much alexgalstyan, very much helped me