platformio/platform-espressif8266 #161: Not able to build anything anymore

TD-er, 2019-06-23:
After upgrading PIO and/or the platformio framework packages (not sure which one caused these issues), I can no longer build anything anymore.
I already removed all framework packages in the .platformio directory, but I cannot compile anything anymore.

Any idea where to look for an option to fix this?

```
platformio run --environment normal_ESP8266_4M
Processing normal_ESP8266_4M (framework: arduino; platform: https://github.com/TD-er/platform-espressif8266.git#patch/v1.8.0_Puya; board: esp12e)
 
Error: Traceback (most recent call last):
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\__main__.py", line 88, in main
    cli(None, None, None)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__

    return self.main(*args, **kwargs)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 37, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 99, in cli
    result = (envname, ep.process())
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 61, in process
    result = self._run_platform()
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 115, in _run_platform
    return p.run(build_vars, build_targets, self.silent, self.verbose)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 373, in run
    assert isinstance(targets, list)
AssertionError
 
============================================================
 
An unexpected error occurred. Further steps:
 
* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command
 
* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html
 
* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues
 
============================================================
```

alexgalstyan, 2019-06-27:
I have same problem to:
[built results.txt](https://github.com/platformio/platform-espressif8266/files/3334837/built.results.txt)
What happened ?!

alexgalstyan, 2019-06-27:
I have found the problem

ivankravets, 2019-06-27:
What is your Python core version?