platformio/platform-espressif8266 #162: Not able to build anything anymore

TD-er, 2019-06-28:
After upgrading PIO and/or the platformio framework packages (not sure which one caused these issues), I can no longer build anything anymore.
I already removed all framework packages in the .platformio directory, but I cannot compile anything anymore.

Any idea where to look for an option to fix this?

```
platformio run --environment normal_ESP8266_4M
Processing normal_ESP8266_4M (framework: arduino; platform: https://github.com/TD-er/platform-espressif8266.git#patch/v1.8.0_Puya; board: esp12e)
 
Error: Traceback (most recent call last):
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\__main__.py", line 88, in main
    cli(None, None, None)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__

    return self.main(*args, **kwargs)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 37, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 99, in cli
    result = (envname, ep.process())
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 61, in process
    result = self._run_platform()
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 115, in _run_platform
    return p.run(build_vars, build_targets, self.silent, self.verbose)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 373, in run
    assert isinstance(targets, list)
AssertionError
 
============================================================
 
An unexpected error occurred. Further steps:
 
* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command
 
* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html
 
* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues
 
============================================================
```

mcspr, 2019-07-01:
@TD-er I tried building ESPEasy with the PIO4 some time ago, it seems there is something from with the platformio.ini config. Couple of issues that I found just now:
https://github.com/letscontrolit/ESPEasy/blob/a8ab3bcd7c5ea24c85abd48a64a2d24fa8045f81/platformio.ini#L178
This causes the assert error. Removing that seems clear the run part, but it still halts when parsing build_flags:
```python
IndexError: string index out of range:
  File "/home/builder/packages/platformio-core/platformio/builder/main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/builder/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/builder/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/builder/.platformio/platforms/espressif8266@2.2.0/builder/main.py", line 203:
    target_elf = env.BuildProgram()
  File "/home/builder/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/home/builder/packages/platformio-core/platformio/builder/tools/platformio.py", line 109:
    env.ProcessFlags(env.get("BUILD_FLAGS"))
  File "/home/builder/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/home/builder/packages/platformio-core/platformio/builder/tools/platformio.py", line 191:
    env.Append(**env.ParseFlagsExtended(flags))
  File "/home/builder/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/home/builder/packages/platformio-core/platformio/builder/tools/platformio.py", line 154:
    for key, value in env.ParseFlags(str(raw)).items():
  File "/home/builder/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 800:
    do_parse(arg)
  File "/home/builder/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 729:
    elif not arg[0] in ['-', '+']:
```

Which made less sense than the previous message, but it looks like build_flags concatenation is the problem. I tried printing what ParseFlags receives:
```
build_flags => ['-DBUILD_GIT=\'""\'', '-DNDEBUG', '-DVTABLES_IN_FLASH', '-fno-exceptions', '-DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH', '-DWEBSERVER_RULES_DEBUG=0', '-DPUYA_SUPPORT=1', '-DCORE_POST_2_5_0 "" -Wl,-Tesp8266.flash.2m256.ld -D PLUGIN_SET_SHELLY_1']
```
Notice the `""` between `-DCORE_POST_2_5_0` and `-Wl,-T`. Not quite sure where that came from

TD-er, 2019-07-02:
@ivankravets Not sure why I did not see your update in the past days.
At the time of reporting the issue, it was the last one available.
Since then I did not use my laptop downstairs to build anymore and did not update stuff on my other computer.

@mcspr Thanks, I will test it. (and fix it)

ivankravets, 2019-07-02:
It's very difficult to reproduce if "something does not build". Please provide a simple project to reproduce this issue.

mcspr, 2019-07-02:
Regarding the `targets = ` in config, it does fail here because it does not return the specified `default=[]` 
https://github.com/platformio/platformio-core/blob/0ce23438368e4d9fa323649c0ad7f9027ce26938/platformio/commands/run/processor.py#L87
https://github.com/platformio/platformio-core/blame/0ce23438368e4d9fa323649c0ad7f9027ce26938/platformio/project/config.py#L207

Technically config *does* specify it, but empty string is not treated as list. So I am not sure whether this is an issue or not. Error traceback does look scarrer for what is really happenig
https://github.com/platformio/platformio-core/blame/0ce23438368e4d9fa323649c0ad7f9027ce26938/platformio/project/options.py#L93

@ivankravets 
Here's minimal ini test (this should be added to tests?):
```
[env:d1_mini]
platform = espressif32
board = lolin32
framework = arduino
targets =
```