platformio/platform-espressif8266 #164: Linking fails on feature/stage branch

Niek, 2019-07-10:
The develop and old versions work fine, but when trying to build my project on the feature/stage branch I get this error from the linker:
```bash
~/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/my-env/libFrameworkArduino.a(Esp.cpp.o): in function `EspClass::getSketchSize()':
~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/Esp.cpp:520: undefined reference to `_FS_start'
collect2: error: ld returned 1 exit status
*** [.pio/build/my-env/firmware.elf] Error 1
```

This seems to be related to this PR that was merged a month ago: https://github.com/esp8266/Arduino/pull/5511

Niek, 2019-07-11:
It seems the toolchain in ``feature/stage`` is at an old version: 2.40802.190218 (4.8.2). To build it needs a toolchain of at least 3.0: https://github.com/earlephilhower/esp-quick-toolchain/releases

mcspr, 2019-07-13:
3.0... looks like a testing branch for gcc-7 (and even gcc-9 as of right now!)

_FS_start is expected to be provided by the linker script:
https://github.com/esp8266/Arduino/blob/8b5433977edbe9749d3d7cdfc6a2fb7b19944095/tools/sdk/ld/eagle.flash.1m.ld#L17
Replacing old _SPIFFS_start, _end, _page and _block
Which could be an issue when the used ldscript does not come from the tools/sdk/ld but from the project (using old variables still, for example)

Niek, 2019-07-13:
You're right @mcspr - I needed a change in the linker script, it actually builds fine with the old toolchain - so closing this.

closed 2019-07-13 by Niek