platformio/platform-espressif8266 #165: Error: Could not find LD script  when uploading remotely

vrabcak, 2019-07-10:
I have following setup: Platformio installed on Raspberry Pi and gen4iod board connected to the Raspberry via USB. I run remote agent on the Raspberry. 

When I try to upload firmware remotely from my PC, it ends with the error: 

```
================================================================== [SUCCESS] Took 6.25 seconds ==================================================================
Uploading firmware remotely
Processing gen4iod (platform: espressif8266; board: gen4iod; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Error: Could not find LD script
=================================================================== [ERROR] Took 1.79 seconds ===================================================================
The terminal process terminated with exit code: 1
```

When I compile exactly the same project locally on the Raspberry and upload it locally (with `pio run --target upload`), it uploads the firmware to the board without any problems.

mcspr, 2019-07-11:
I'm seeing the same error when specifying `pio run -e env -t nobuild -t upload`

After some tracing, I noticed that the error message comes from platformio-core:
https://github.com/platformio/platformio-core/blob/5e73348263ef10458a950a0022dabaec803f46a6/platformio/builder/tools/piomisc.py#L235-L277

But `LINKFLAGS` is missing. Only place that I've noticed it may come from is target elf generation, but it will only trigger on building:
https://github.com/platformio/platformio-core/blob/5e73348263ef10458a950a0022dabaec803f46a6/platformio/builder/tools/platformio.py#L109

https://github.com/platformio/platform-espressif8266/blob/fa48bca577365b9cf1870fe3a443b02419874ff2/builder/main.py#L204-L205
Adding the `env.ProcessFlags(env.get("BUILD_FLAGS"))` fixes the LD script detection.
But it looks like that the platformio-core code must always parse build flags and not the platform

mariusstaicu, 2019-07-11:
Same problem using espressif8266@2.2.2 with platformio 3.6.7. 
However, quick workaround suggested here  https://community.platformio.org/t/esp8266-error-on-remote-uploading/4618 works. Adding --force-remote solves the problem by building on the remote agent.

liebman, 2019-07-17:
The workaround is not working for me with pio v4.0.0.

lakidd, 2019-07-21:
For me, the workaround leads to 

```
/home/lakidd/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/d1_mini_pro/libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
/home/lakidd/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/d1_mini_pro/libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
/home/lakidd/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/d1_mini_pro/libFrameworkArduino.a(core_esp8266_main.cpp.o): in function `loop_wrapper()':
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x21): undefined reference to `setup'
/home/lakidd/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x2d): undefined reference to `loop'
collect2: error: ld returned 1 exit status
*** [.pio/build/d1_mini_pro/firmware.elf] Error 1
```

mariusstaicu, 2019-07-30:
for me is not an option to use --force-remote, I need to build on CI server

ivankravets, 2019-09-02:
Please re-run `pio remote update` or `pio update` on the remote device. Fixed in https://github.com/platformio/platform-espressif8266/releases/tag/v2.2.3

closed 2019-09-02 by ivankravets

lakidd, 2019-09-03:
Yep, that fixed it ! Thanks

mariusstaicu, 2019-10-05:
fixed, thanks

commit e69a2cb6 (2019-09-02): Fix an issue with "Could not find LD script" when uploading via PIO Remote // Resolve #165