platformio/platform-espressif8266 #167: Update toolchain to 2.5.0-4

Niek, 2019-07-23:
See https://github.com/earlephilhower/esp-quick-toolchain/releases/tag/2.5.0-4

Current version of the toolchain is "2.40802.190218" which is actually 2.5.0-3. It would be nice if the toolchain could be updated to the latest stable version which is 2.5.0-4.

Niek, 2019-09-30:
There are also test releases out for 3.0. In case anybody is looking to test it (macOS example):
```bash
cd ~/.platformio/packages/
curl https://github.com/earlephilhower/esp-quick-toolchain/releases/download/3.0.0-gnu2/x86_64-apple-darwin14.xtensa-lx106-elf-dd9f9a2.1569802152.tar.gz | tar xvfz
mv xtensa-lx106-elf toolchain-xtensa@3
cat <<EOF > toolchain-xtensa@3/package.json
{
    "name": "toolchain-xtensa",
    "system": [
        "darwin_x86_64",
        "darwin_i386"
    ],
    "url": "https://github.com/earlephilhower/esp-quick-toolchain",
    "version": "3.0.0"
}
EOF
```
Then, edit your ``platformio.ini`` and add ``platform_packages = toolchain-xtensa@3``

mcspr, 2019-09-30:
As I understand it, 2.5.0-4 should be part of the Core 2.6.0 release, so perhaps it would make sense to upgrade feature/stage branch to use it there. *If* that is still relevant with the platform_packages = ... addition, but user needs to manually add manifest file like it is described above

@Niek could you please share this approach in the gcc-9 PR for other platforms?
https://github.com/earlephilhower/esp-quick-toolchain/releases/
https://github.com/esp8266/Arduino/pull/6294

Jason2866, 2019-10-20:
+ 1 for using toolchain 2.5.0-4 in feature/stage. It is used as standard in Arduino stage version

Jason2866, 2019-11-08:
It is needed now. See https://github.com/platformio/platform-espressif8266/issues/180

Niek, 2019-11-09:
Core 2.6.0 has been released now (https://github.com/esp8266/Arduino/releases/tag/2.6.0) and it requires the new toolchain in order to build binaries that do not disconnect regularly.

Niek, 2019-11-14:
2.6.1 is out now: https://github.com/esp8266/Arduino/releases/tag/2.6.1

Any update?

closed 2019-11-25 by valeros

Jason2866, 2019-11-25:
The entry for build chain 2.5.04 is missing in https://dl.bintray.com/platformio/dl-packages/manifest.json

commit aebb1c40 (2019-11-25): Update Arduino core to 2.6.1 // Resolve #181, resolve #180, resolve #167