platformio/platform-espressif8266 #174: build_cache_dir mostly ignored

neophiliac, 2019-09-13:
I'm building with both the IDE and the CLI, and sometimes I rarely see output about getting objects from the cache while building the arduino framework files for esp8266, but most of the time all of the framework files are recompiled every time. I have build_cache_dir set (and also PLATFORMIO_BUILD_CACHE_DIR) and the build_cache_dir subdirectories are populated with object files with every build.

The build log shows the object files being put in a local directory, e.g. .pio/build/debug_ota/FrameworkArduino/spiffs/spiffs_check.cpp.o (which is not the build_cache_dir). The files in the build_cache_dir directories have hashes for names. and new files are added to each subdirectory with each build.

I have run pio update && pio upgrade so everything is up to date, PlatformIO 4.0.3 on Ubuntu 19.04.

It seems like I must be missing something about using the build cache setting. Suggestions? Advice?

Thanks.

platformio.ini:
```
[platformio]
default_envs = debug_ota
build_cache_dir = /tmp/build_cache

[env]
platform = espressif8266
lib_extra_dirs = ~/Documents/Arduino/libraries
board = d1_mini
board_build.flash_mode = dout
framework = arduino
lib_deps=
  ArduinoJson@~5.13
  NtpClient
monitor_speed = 115200
monitor_baud = 115200
upload_speed = 256000

[env:production]
build_flags = 
  !echo "-DGIT_VERSION="$(git describe --tags --long)" -DBUILD_DATE="$(date +'%y%m%d-%H%M%S')" -DGIT_BUILDER="$(git config user.email)
  -Wl,-Map,linker.map
  -DDEBUG_ESP_PORT=Serial

[env:debug_ota]
build_flags = 
  !echo "-DGIT_VERSION="$(git describe --tags --long)" -DBUILD_DATE="$(date +'%y%m%d-%H%M%S')" -DGIT_BUILDER="$(git config user.email)
  -Wl,-Map,linker.map
  -DDEBUG_ESP_PORT=Serial
  -DDEBUG_ESP_SSL
  -DDEBUG_ESP_UPDATER
  -DDEBUG_ESP_OTA

```

mcspr, 2019-10-21:
See https://community.platformio.org/t/build-cache-dir-will-not-share-object-files-between-envs/10011
in the example above, you modify global build_flags = ... so those flags are appended to the compiler cmd when building framework, framework libs and lib_deps, thus modifying the hash under which cache dir stores the resulting objects.
as far as I understood it, [at this time different envs will not share build cache](https://community.platformio.org/t/build-cache-dir-will-not-share-object-files-between-envs/10011/8?u=mcspr) because of the same thing

maybe this issue should be moved to [platformio/platformio-core](https://github.com/platformio/platformio-core/issues)?

neophiliac, 2019-11-02:
Yes, it's probably a platformio-core issue. Sorry for leaving this in the wrong place. I'll read the docs you linked before I do that though. 

Thanks.

closed 2019-11-02 by neophiliac

LordMike, 2023-03-17:
For future readers. I've made this issue with platformio: https://github.com/platformio/platformio-core/issues/4574

ivankravets, 2023-03-18:
> -DBUILD_DATE="$(date +'%y%m%d-%H%M%S')"

This is the issue that triggers recompilation. The macro value changes on each build.

LordMike, 2023-03-18:
Ooh. Ignore my interpretation then for this issue. :)