platformio/platform-espressif8266 #176: Option to share build folder based on used platform

TD-er, 2019-09-15:
I do experience lots of Travis build timeouts for my project. (ESPeasy)
I do have a single platformio.ini file which has a lot of build environments defined which have their own sets of plugins defined.
On top of those preset defines, the used platform also differs.
For example ESP8266/Arduino with some variety in core versions, and ESP32/Arduino.

Would it be possible (and if not, this is a feature request) to define a build (or cache) dir to keep the .o files based on the active platform?
For example `espressif8266@2.2.3` in one dir and `espressif32@1.9.0` in the other, etc.

mcspr, 2019-10-21:
Maybe related to the https://github.com/platformio/platform-espressif8266/issues/174#issuecomment-544431763 (and the linked pio-community thread)?

LordMike, 2023-03-17:
For future readers. I've made this issue with platformio: https://github.com/platformio/platformio-core/issues/4574