platformio/platform-espressif8266 #178: Can't get serial data from Serial monitor

morestart, 2019-10-02:
this is my test code
```c
void setup(){
  Serial.begin(9600);
}

void loop(){
  Serial.println("OK");
  delay(1000);
}
```

serial monitor always no data, i use vs code & win10 1903. But when i use serial tools, i can recive the serial data.

this is my platform.ini content:

```
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
monitor_speed = 9600
upload_speed = 961200
```

pfeerick, 2019-11-26:
Double check the correct serial port is being used. For example, your system may have a built-in serial port, and PlatformIO connecting to that by mistake.

morestart, 2019-11-26:
> Double check the correct serial port is being used. For example, your system may have a built-in serial port, and PlatformIO connecting to that by mistake.

yeah,you are right

closed 2019-11-26 by morestart