platformio/platform-espressif8266 #179: flashing loop

zen29d, 2019-11-07:
Board: NodeMCU
framework: arduino

when i tried to upload blink [.platformio/platforms/espressif8266/examples/arduino-blink], it upload same program many time

 '
No dependencies
Compiling .pio/build/d1_mini/src/main.cpp.o
Generating LD script .pio/build/d1_mini/ld/local.eagle.app.v6.common.ld
Archiving .pio/build/d1_mini/libFrameworkArduinoVariant.a
Indexing .pio/build/d1_mini/libFrameworkArduinoVariant.a
Compiling .pio/build/d1_mini/FrameworkArduino/Esp-frag.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/Esp-version.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/FS.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/MD5Builder.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/Print.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/Schedule.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/ScheduledFunctions.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/StackThunk.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/StreamString.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/Updater.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/WString.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/abi.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/base64.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/cbuf.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/cont.S.o
Compiling .pio/build/d1_mini/FrameworkArduino/cont_util.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_app_entry_noextra4k.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_eboot_command.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_flash_utils.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_i2s.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_main.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_noniso.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_phy.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_postmortem.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_si2c.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_sigma_delta.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_timer.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_waveform.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_wiring.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_wiring_analog.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_wiring_digital.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_wiring_pulse.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_wiring_pwm.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_wiring_shift.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/debug.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/gdb_hooks.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/heap.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/libb64/cdecode.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/libb64/cencode.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/libc_replacements.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/sntp-lwip2.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/spiffs/spiffs_cache.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/spiffs/spiffs_check.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/spiffs/spiffs_gc.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/spiffs/spiffs_hydrogen.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/spiffs/spiffs_nucleus.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/spiffs_api.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/spiffs_hal.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/sqrt32.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/time.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/uart.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/umm_malloc/umm_malloc.cpp.o
[... 917 lines cut ...]
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pio/build/thing/src/main.cpp.o
Generating LD script .pio/build/thing/ld/local.eagle.app.v6.common.ld
Archiving .pio/build/thing/libFrameworkArduinoVariant.a
Indexing .pio/build/thing/libFrameworkArduinoVariant.a
Compiling .pio/build/thing/FrameworkArduino/Esp-frag.cpp.o
Compiling .pio/build/thing/FrameworkArduino/Esp-version.cpp.o
Compiling .pio/build/thing/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/thing/FrameworkArduino/FS.cpp.o
Compiling .pio/build/thing/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pio/build/thing/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/thing/FrameworkArduino/IPAddress.cpp.o
^C*** [.pio/build/thing/FrameworkArduino/FunctionalInterrupt.cpp.o] Build interrupted.
*** [.pio/build/thing/FrameworkArduino/FS.cpp.o] Build interrupted.
*** [.pio/build/thing/FrameworkArduino/HardwareSerial.cpp.o] Build interrupted.
*** [.pio/build/thing/FrameworkArduino/IPAddress.cpp.o] Build interrupted.
Error: Aborted by user
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
'

pfeerick, 2019-11-26:
Look at the end of the header where it says `board`

e.g. 
```
Processing thing (platform: espressif8266; framework: arduino; board: thing)
------------------------------------------------------------------------------
```

Now look at all the other instances... each time a different board is being targeted. You either need to add [default_envs](https://docs.platformio.org/en/latest/projectconf/section_platformio.html#default-envs) to your projects `platformio.ini` and set it to the board/environment you want to compile when you hit the 'build (all)' button, or choose the specific environment you want to build.

zen29d, 2019-11-28:
I've set the environment for the right board, in fact, it flashes it before, and after this event it never occur again. I don't know what happed then but now its work.

closed 2019-11-28 by zen29d