platformio/platform-espressif8266 #180: ESP8266 SDK22x_191024 does not work reliable with old Xtensa Build chain

Jason2866, 2019-11-08:
- [x] **PlatformIO Core**.
      
Please update Xtensa build chain for ESP8266 to latest 2.5.x version for feature/stage.
The newer Non Os SDK versions 2.2.2 from 191024 and 191105 does NOT work
reliable with the the used build chain.

When compiling Tasmota with the Xtensa Build Chain 2.4.x there are wifi reconnects.
Using Xtensa Build chain 2.5.0.4 solves this issues.

I have done a custom setup (dirty hack for windows) to use the 2.5.x version.
https://github.com/Jason2866/platform-espressif8266.git#SDK22x_191105
   

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 pro 

**PlatformIO Version** (`platformio --version`): Latest dev version

### Description of problem
Compiled firmware is unreliable.

#### Steps to Reproduce
Compile Tasmota with old Xtensa Build chain.
Issue solving:
Compile Tasmota with new (in Arduino IDE default) build chain

**The content of `platformio.ini`:**

[platformio.zip](https://github.com/platformio/platformio-core/files/3824513/platformio.zip)

**Source file to reproduce issue:**
https://github.com/arendst/Tasmota

Jason2866, 2019-11-08:
Arduino Core 2.6.0 will be released this week :-)
https://github.com/esp8266/Arduino/pull/6724#issuecomment-551610434

TD-er, 2019-11-08:
> Arduino Core 2.6.0 will be released this week :-)
> [esp8266/Arduino#6724 (comment)](https://github.com/esp8266/Arduino/pull/6724#issuecomment-551610434)

It has been released :)

ascillato, 2019-11-09:
@ivankravets 

Hi,

Core 2.6.0 has been just released but requires to update toolchain to 2.5.0-4

Using previous toolchain (2.4.x) produces firmwares that have wifi disconnections for ESP8266 devices.

Thanks.

Jason2866, 2019-11-14:
Core 2.6.1 is released
@TD-er  You can use with 
`platform                  = https://github.com/Jason2866/platform-espressif8266.git#core_2_6_1`

dirkmueller, 2019-11-15:
Duplicate of #167

arendst, 2019-11-17:
Hi @ivankravets pls give this priority as I would like to release Tasmota one of these days using a stable platformio environment

Thx in advance.

ivankravets, 2019-11-17:
Could someone summarize what should be done from our side? We have support for Arduino and native SDK. Is the new compiler compatible with SDK version which we support? Or, the new toolchain should be used for new Arduino?

Jason2866, 2019-11-17:
@ivankravets  **Release core 2.6.1 as platformio-espressif8266 package with Xtensa build chain 2.5.0.4**
The new SDK version SDK22x_191105 does solve many issues but requires Xtensa build chain 2.5.0.4.
If the older build chain is used it generates wifi issues

ivankravets, 2019-11-17:
@Jason2866 I mean "framework-esp8266-rtos-sdk" and "framework-esp8266-nonos-sdk". Are they compatible with this new toolchain?

Jason2866, 2019-11-17:
@ivankravets  tbh, i dont know. Never used framework-esp8266-rtos-sdk
Arduino ESP8266 is only nonos-sdk.

dirkmueller, 2019-11-17:
@ivankravets as far as I can see it is just bugfixes, should be compatible with anything

Jason2866, 2019-11-23:
@ivankravets  any news? Maybe a release date? ;-)

closed 2019-11-25 by valeros

Jason2866, 2019-11-25:
@valeros  Thx, but it does still install old Xtensa toolchain.
[image]

valeros, 2019-11-25:
`2.40802.191122` it's our internal version of `2.5.0-4`

Jason2866, 2019-11-25:
Ahh thx!

arendst, 2019-11-25:
Thx too.

TD-er, 2019-11-25:
Thanks three :)

andrethomas, 2019-11-25:
And four, much appreciated :)

commit aebb1c40 (2019-11-25): Update Arduino core to 2.6.1 // Resolve #181, resolve #180, resolve #167