platformio/platform-espressif8266 #181: ESp8266 Arduino V2.6.0

mars000, 2019-11-09:
I noticed this version just released on GitHub.
Just for my understanding does PlatformIO automatically pick this up as an updated within the PlatformIO GUI in VSC or do I need to wait for someone to publish a PlatformIO file of sorts ?

Jason2866, 2019-11-09:
See #180

Niek, 2019-11-11:
Also: https://github.com/platformio/platform-espressif8266/issues/167

closed 2019-11-25 by valeros

Niek, 2019-11-25:
Thanks @valeros, are you going to release a version on GitHub as well?

valeros, 2019-11-25:
If there are no critical issues, it'll be released within a few days. As for now you can try the upstream vesion:
```
[env:upstream_develop]
platform = https://github.com/platformio/platform-espressif8266.git
board = ...
```

dirkmueller, 2019-11-25:
It looks like tool-esptoolpy is still old, it should be 1.20800.0, with other words it needs to include https://github.com/espressif/esptool/releases/tag/v2.8

valeros, 2019-11-25:
Do you experience any issues with the current `tool-esptoolpy`?

dirkmueller, 2019-11-25:
@valeros not yet, however this is not matching the official release.. when you download https://github.com/esp8266/Arduino/releases/download/2.6.1/esp8266-2.6.1.zip

and do

```
$ grep __version__ tools/esptool/esptool.py
__version__ = "2.8"
```

you can see that it has 2.8, while platformio installs 2.6:

```
grep __version__ ~/.platformio/packages//tool-esptoolpy/esptool.py 
__version__ = "2.6"
```

commit aebb1c40 (2019-11-25): Update Arduino core to 2.6.1 // Resolve #181, resolve #180, resolve #167