platformio/platform-espressif8266 #183: Cannot get WiFi working with Arduino framework

jrbaxter0, 2019-11-19:
My long term goal is to this: https://github.com/dogmaphobic/mavesp8266 working on my ESP-01 (1MB) device. However, I cannot get it to work, and started using some less complex examples to try to debug the issue.

First, I can get Blink and the WifiAccessPoint examples working when I compile and upload them via the Arduino IDE. I am using an FTDI adapter on a Windows machine.

Next, I was able to get Blink working after I manually set the pin to be 2 (rather than using the macro, which did not work). I am using esp01_1m, compliled with `platformio run`, then uploaded with `esptool.py --baud 921600 --port /dev/ttyUSB0 write_flash 0x00000 .pio/build/esp01_1m/firmware.bin` This was done on an 18.04 Ubuntu VM.

So, I believe I've verified the hardware works, and that my process works under my Linux VM. However, when I run the stock Arduino WiFi access point example, I can't get it to work. There is no broadcasted SSID. 

Do you have ideas on steps I can take to troubleshoot or workarounds?

main.cpp
```
/*
   Copyright (c) 2015, Majenko Technologies
   All rights reserved.

   Redistribution and use in source and binary forms, with or without modification,
   are permitted provided that the following conditions are met:

 * * Redistributions of source code must retain the above copyright notice, this
     list of conditions and the following disclaimer.

 * * Redistributions in binary form must reproduce the above copyright notice, this
     list of conditions and the following disclaimer in the documentation and/or
     other materials provided with the distribution.

 * * Neither the name of Majenko Technologies nor the names of its
     contributors may be used to endorse or promote products derived from
     this software without specific prior written permission.

   THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND
   ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED
   WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
   DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE FOR
   ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES
   (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
   LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON
   ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
   (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
   SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
*/

/* Create a WiFi access point and provide a web server on it. */

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

#ifndef APSSID
#define APSSID "ESPap"
#define APPSK  "thereisnospoon"
#endif

/* Set these to your desired credentials. */
const char *ssid = APSSID;
const char *password = APPSK;

ESP8266WebServer server(80);

/* Just a little test message.  Go to http://192.168.4.1 in a web browser
   connected to this access point to see it.
*/
void handleRoot() {
  server.send(200, "text/html", "<h1>You are connected</h1>");
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  Serial.print("Configuring access point...");
  /* You can remove the password parameter if you want the AP to be open. */
  WiFi.softAP(ssid, password);

  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  server.on("/", handleRoot);
  server.begin();
  Serial.println("HTTP server started");
}

void loop() {
  server.handleClient();
}
```

Jason2866, 2019-11-23:
@jrbaxter0  github issues is for problems with the software itself.
It is not a general forum. Please ask for help in forums which are made for.

jrbaxter0, 2019-11-24:
@Jason2866 I believe this is an issue with the software, not my code. I have posted a section of example code, which I have not even written. It compiles and runs on the board through the Arduino IDE. When using platformIO (on both Windows and Linux with the same physical setup as Arduino IDE), the ESP01 board does not generate a WiFi signal.

I have found an alternate solution for my problem, which does not require platformIO, but this issue (which I believe is in the software, not my setup in particular) is not resolved.

valeros, 2019-11-25:
Hi @jrbaxter0! Could you please try the upstream vesion?
```
[env:upstream_develop]
platform = https://github.com/platformio/platform-espressif8266.git
board = ..
```

valeros, 2020-05-04:
Please reopen if you still need help.

closed 2020-05-04 by valeros