platformio/platform-espressif8266 #186: Update to core 2.6.2

Niek, 2019-11-28:
Core 2.6.2 has been released, this is a bugfix release: https://github.com/esp8266/Arduino/releases/tag/2.6.2 that fixes some issues (among which WiFi issues).

ivankravets, 2019-11-28:
Please update to 2.3.1 https://github.com/platformio/platform-espressif8266/releases/tag/v2.3.1

closed 2019-11-28 by ivankravets

Niek, 2019-11-28:
Thanks for the quick release!

Pablo2048, 2019-11-28:
Thank you too. Unfortunately it seems like there is not the last gcc. According to this:

`PLATFORM: Espressif 8266 2.3.1 > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 2.40802.191122 (4.8.2), framework-arduinoespressif8266 2.20602.191128 (2.6.2), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20800.0 (2.8.0`

It seems like xtensa toolchain 4.8.2. But when I try to use @Jason2866 2.6.1 pre-release it shows:

`PLATFORM: Espressif 8266 2.2.4 #043f5fd > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 2.50400.170719 (5.4.0), framework-arduinoespressif8266 2.20601.191124 (2.6.1), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20800.0 (2.8.0)`

which is 5.4.0, but with 2017 year so I'm confused. Can anybody please put some light to this? Thanks.

ivankravets, 2019-11-28:
@valeros, please help

ascillato, 2019-11-28:
@Pablo2048 

toolchain-xtensa 2.40802.191122 (4.8.2) for platformio standard IS THE SAME as toolchain-xtensa 2.50400.170719 (5.4.0)

Pablo2048, 2019-11-28:
Ok. Thank you for explanation.

commit 95bfa9a4 (2019-11-28): Update Arduino core to 2.6.2 // Resolve #186