platformio/platform-espressif8266 #191: ESP8266: resetmethod is being ignored

middelink, 2020-01-23:
Looking at the code in `platforms/expressif8266/builder/main.py` I see the ERASECMD/UPLOADCMD always using `esptool.py` which has no support for either `ck` or `wifio` modes, nor does it add the reset method as a flag to the uploader.

Possible solutions I see:
a) remove the documentation stating `resetmethod` exists and from all `boards.txt`. Leaving a few people like me stranded...
b) switch to `esptool-ck` tool for these 2 modes, which would require quite a lot of refactoring in `main.py`
c) add the resetmethods to `esptoo.py` and add `resetmethod` as a flag to upload/erase?
d) switch to `github.com/igrr/esptool-ck` and add the `resetmethod` as a flag to upload/erase?

I'm unsure about c) or d) - your `esptool-ck` seem out of date? If we go for c) I can send you a PR implementing the other reset methods. Lemme know what you think.

condemil, 2020-01-28:
c) https://github.com/espressif/esptool/pull/508/files

middelink, 2020-01-28:
Yeah, that was me. Not much response though.

closed 2020-05-04 by valeros

middelink, 2020-05-04:
So what is the resolution here?

valeros, 2020-05-05:
@middelink Now `ck` and 'nodemcu' reset methods are converted to proper `esptool.py` flags, so you can use them as before.

commit 5c75ddb0 (2020-05-04): Handle resetmethod field when esptoolpy is used as the uploader
Resolve #191