platformio/platform-espressif8266 #193: [upload] Error 2 with esp8266 ESP-07 board

daryooosh, 2020-01-28:
I use VSCODE to create project. I use a project and select `ESp-07 board` and `ESP8266 RTOS SDK` framework then create a `main.c` file in `src` directory and copy content of `main.c` file of [`esp8266-rtos-sdk-blink`](https://github.com/platformio/platform-espressif8266/blob/master/examples/esp8266-rtos-sdk-blink/src/main.c) example to my project `main.c` file. it compile successfully but when i want to upload it to my board following error occur :

> Processing esp07 (platform: espressif8266; framework: esp8266-rtos-sdk; board: esp07)
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp07.html
PLATFORM: Espressif 8266 2.3.2 > Espressif Generic ESP8266 ESP-07
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), framework-esp8266-rtos-sdk 1.5.0-beta.5, tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20800.0 (2.8.0), tool-mkspiffs 1.200.0 (2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Retrieving maximum program size .pio\build\esp07\firmware.elf
Checking size .pio\build\esp07\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [====      ]  36.3% (used 29754 bytes from 81920 bytes)
PROGRAM: [=======   ]  73.9% (used 278664 bytes from 376832 bytes)
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM3
Uploading .pio\build\esp07\firmware.bin
esptool.py v2.8
Serial port COM3
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:33:a6:e6
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
A fatal error occurred: File C:\Users\MrTavakoli\.platformio\packages\framework-esp8266-rtos-sdk\bin\esp_init_data_default.bin (length 128) at offset 4177920 will not fit in 1048576 bytes of flash. Use --flash-size argument, or change flashing address.
*** [upload] Error 2

I test my board with `Arduino` framework successfully.

daryooosh, 2020-01-29:
I use esp01-1M board instead of esp07 board and compile then upload to esp07 board successfully.
again change board to esp07 and add

board_upload.maximum_size = 1048576

to platformio.ini file and compile and upload toesp07` board successfully.
I found 1048576 value from esp01_1m.json file.

gonzabrusco, 2020-11-26:
Could somebody please merge that pull request? I can confirm the current configuration for esp07 does not work. The esp07 is autodetected as 1M.

PR #204 (2020-11-30): Fix ESP-07 flash size
Should fix #193. The ESP-07 has only 1MB of flash, it’s the ESP-07S which has 4MB.

I couldn't read the EEPROM because it was defined outside the flash.
commit 468a6250 (2020-11-30): Fix ESP-07 flash size (#204)

Fix ESP-07 flash size and add a new board ESP-07S
Resolve #193
closed 2020-11-30 by valeros