platformio/platform-espressif8266 #196: Unable to specify upload_port with wildcard

egueli, 2020-02-06:
Expected:
1. Initialize a PlatformIO project
2. In platformio.ini, set `upload_port` to a path with wildcards, e.g. `/dev/tty.wchusbserial*`
3. run `pio run -t upload`
4. wait for the firmware to be compiled and uploaded

Actual:
4. the upload fails with the following:
```
Creating BIN file ".pio/build/d1_mini/firmware.bin" using ".pio/build/d1_mini/firmware.elf"
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
```

Contents of platformio.ini
```
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
upload_port = /dev/tty.wchusbserial*
monitor_port = /dev/tty.wchusbserial*
monitor_speed = 115200
```

On Mac I find it particularly important to set the port via wildcards, because the auto-detection doesn't seem to work correctly: it picks a `/dev/cu.usbserial-*` port, that always fails with `Resource busy`.
The workaround is to specify the exact port path, but every time the port changes (e.g. after replugging the device) it has to be changed too, needing a full build.

ivankravets, 2020-02-07:
Please provide an output from `pio device list`

egueli, 2020-02-07:
```
$ pio device list
/dev/cu.Bluetooth-Incoming-Port
-------------------------------
Hardware ID: n/a
Description: n/a

/dev/cu.SamsungM7-COM0
----------------------
Hardware ID: n/a
Description: n/a

/dev/cu.wchusbserial14230
-------------------------
Hardware ID: USB VID:PID=1A86:7523 LOCATION=20-2.3
Description: USB2.0-Serial

/dev/cu.usbserial-14230
-----------------------
Hardware ID: USB VID:PID=1A86:7523 LOCATION=20-2.3
Description: USB2.0-Serial

```

Oh... my bad. I was using `tty` instead of `cu`.

closed 2020-02-07 by egueli