platformio/platform-espressif8266 #203: Stage branch gone?

CRCinAU, 2020-03-26:
So I've been using feature/stage for quite a while to build my ESP8266 stuff on - but this branch is currently gone...

I tried master, it comes up as:
```
SDK:2.2.2-dev(38a443e)/Core:2.6.3=20603000/lwIP:IPv6+STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af
```
If I try to use `platform = https://github.com/platformio/platform-espressif8266.git#develop`, I get:
```
SDK:2.2.2-dev(38a443e)/Core:2.6.3=20603000/lwIP:IPv6+STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af
```

When using this, I can't connect to my https server for updates etc...

An older binary on another device can connect fine - it shows as:
```
SDK:2.2.2-dev(38a443e)/Core:2.7.0-dev=20700000/lwIP:IPv6+STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
```

What's to do?

ivankravets, 2020-03-27:
See docs https://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

closed 2020-03-27 by ivankravets

TD-er, 2020-03-28:
The docs state you still should build using feature/stage.

See [here](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform)

CRCinAU, 2020-03-28:
Yeah - I think this is a recent change... My the new setup works... My platformio.ini is as follows:

```
[env:d1_mini]
board = d1_mini
board_build.ldscript = eagle.flash.4m.ld
board_build.f_cpu = 160000000L
board_build.f_flash = 80000000L
framework = arduino
platform = espressif8266
platform_packages =
    framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git
build_flags =
;  -DDEBUG_ESP_PORT=Serial
;  -DDEBUG_ESP_SSL
;  -DDEBUG_ESP_TLS_MEM
;  -DDEBUG_ESP_HTTP_CLIENT
;  -DDEBUG_ESP_HTTP_SERVER
;  -DDEBUG_ESP_CORE
;  -DDEBUG_ESP_WIFI
;  -DDEBUG_ESP_HTTP_UPDATE
;  -DDEBUG_ESP_UPDATER
;  -DDEBUG_ESP_OTA
;  -D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH
  -DPIO_FRAMEWORK_ARDUINO_LWIP2_IPV6_HIGHER_BANDWIDTH
  -DNDEBUG
```

TD-er, 2020-03-28:
Yep that `platform_packages` definition you have here is the key here.
I have something similar added to make it compile again:

https://github.com/letscontrolit/ESPEasy/blob/4d2eddbcdee64623d6d145bcaedc59e3cc747168/platformio_core_defs.ini#L141-L146