platformio/platform-espressif8266 #206: difference between ESP-01 board and Arduino ESP-01 configurations

HamzaHajeir, 2020-04-30:
Hi

I've noticed a difference in Maximum code size that lies in Arduino boards.txt and platformio/boards.

that lies in "not" subtracting the flash segment from total maximum size :
https://github.com/platformio/platform-espressif8266/blob/develop/boards/esp01_1m.json#L25

Here esp01_1m has ldscript of : eagle.flash.1m256.ld
and maximum_size is set to 1048576.

Where it's specified 761840 here : 
https://github.com/esp8266/Arduino/blob/master/boards.txt#L148

And that applies to all ESP8266 boards.

philbowles, 2020-04-30:
If anybody DOES decide to try to align the build process FULLY and correctly with the esp8266 core for ArduinoIDE, I'd be very happy. Three years down the line we are still incorrectly using PLATFORMIO_<boardname> where (obvs) no such entity exists in the original core, and thus anything / everything that depends on  knowing the CORRECT board name and not the PIO-mangled one will (and DOES)  break when compiled with PIO

WHY does PIO insist on making up its own incompatible names? What's wrong with using the correct one? What would break if PIO called the board by its correct name like everyone else does?
How much does PIO value "ArduinoIDE compatibility"

valeros, 2020-05-04:
Thanks for the report! `maximum_size` field simply shows the total size of flash memory. The way how this memory is partitioned is totally up to an application. You can always change `maximum_size` in your `platformio.ini`, e.g.:
```
platform = ...
framework = ...
board = ...
board_upload.maximum_size = 761840
```

pfeerick, 2020-05-05:
> I've noticed a difference in Maximum code size that lies in Arduino boards.txt and platformio/boards.

Which of the eight possible 'maximum sizes' would you like to be used? :-P The configuration for no file system, OTA only [is the nearest](https://github.com/esp8266/Arduino/blob/83166f948bedff85543b1cb5532b905746744df7/boards.txt#L168) to 1048576, but not quite, coming in at 1023984, and probably takes into account the bootloader, reserved space for rfcal, settings, etc. PlatformIO just leaves it in your hands to choose the maximum size, allowing for the easy over-ride rather than having to mess with the `board.txt` configuration when something doesn't suit. 

> Three years down the line we are still incorrectly using PLATFORMIO_ where (obvs) no such entity exists in the original core, and thus anything / everything that depends on knowing the CORRECT board name and not the PIO-mangled one will (and DOES) break when compiled with PIO

Can you elaborate further on this? Looking at say the [esp01_1m](https://github.com/platformio/platform-espressif8266/blob/develop/boards/esp01_1m.json#L7), [d1_mini](https://github.com/platformio/platform-espressif8266/blob/develop/boards/d1_mini.json#L7) or [nodemcu](https://github.com/platformio/platform-espressif8266/blob/develop/boards/nodemcu.json#L7) configurations... where does `PLATFORMIO_` come into the equation? It says `ARDUINO_ESP8266_ESP01`, `ARDUINO_ESP8266_WEMOS_D1MINI`, `ARDUINO_ESP8266_NODEMCU` ... just like it would if compiled on the Arduino IDE.

HamzaHajeir, 2020-05-05:
> Can you elaborate further on this? Looking at say the [esp01_1m](https://github.com/platformio/platform-espressif8266/blob/develop/boards/esp01_1m.json#L7), [...]

For the sake of this point : I've written a [post ](https://community.platformio.org/t/wrong-board-name-for-arduino-boards/13395/3)in PIO community explains it well :