platformio/platform-espressif8266 #207: ESP8266_Exception_decoder filter does not work as expected

geekbozu, 2020-05-04:
When monitoring serial output with the filter decoder active I would expect the stack trace to get re-written
Instead I get an output saying it can not find the .elf file for the build. 
Looking harder it seems that the build directory is getting cleared on calling of monitor. 

Not sure how to fix this, I tried forcing the elf file to read only as a work around which silenced the error but still got no updated stack trace information.

stavultras, 2020-06-04:
For me it does nothing at all. My settings:

```
build_type = debug
monitor_filters = esp8266_exception_decoder
```

It just outputs the regular ESP output, at the same time, when I use the output in Arduino -> ESP Exception Decoder, it decodes the data properly.

sblantipodi, 2020-08-08:
still using the 
monitor_filters = esp8266_exception_decoder, colorize
but it does not work.

nothing changed with the exception decoder or without.

xsrf, 2021-02-14:
Doesn't work for me either with `espressif8266@2.6.3` - output doesn't change 🤷‍♂️

xsrf, 2021-02-16:
Okay, it actually does work, but I ran into two issues:
- whenever you change `platformio.ini`, e.g. to edit `monitor_filters` the firmware/elf binary is deleted from the `.pio\build` directory. Without the binary the filter will just do nothing. The binary must be present the moment monitoring is started! So one has to **rebuild and restart monitor after adding the filter**!
- It also won't work together with the `time` filter

stavultras, 2021-02-16:
@xsrf I tried to rebuild it many times without success. The binary file was there and nothing was happening. Have no idea why.

GOB52, 2021-05-22:
@sblantipodi  @xsrf 

> It also won't work together with the time or colorize filter

If you change the order of the monitor_filters, it might work.
see also https://github.com/platformio/platform-espressif32/issues/545

sblantipodi, 2021-05-22:
it worked for me by simply adding 
build_type = debug

saikek, 2021-07-13:
You need to remember to compile / upload sketch before you debug, since it's using .elf file for mapping issue.
Also, as mentioned:
```
[platformio]
build_type = debug
monitor_filters = esp8266_exception_decoder
```