platformio/platform-espressif8266 #208: ESP8266 ESP-201 What  Configuration .ini ??

rserroyen, 2020-05-04:
Hello
I am trying to use an ESP8266 ESP-201. What configuration platform, board, build_flag, board_build.ldscript, ....

When uploading the flash (.html, .js, .css) and cpp everything goes well but it crashes

Thank you

` ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld

Exception (28):
epc1=0x4020aae5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffca0 end: 3fffffc0 offset: 01a0
3ffffe40:  3ffffe80 00000000 3fff12d4 3fff104c
3ffffe50:  3fff1044 3ffe8962 3ffffec8 4020ac9d  
3ffffe60:  000000f0 3fff104c 3fff1044 4020ae8d
3ffffe70:  00000008 00000000 61000a0d 3ffe8962  
3ffffe80:  3ffef600 3ffef600 3ffffec0 4020802a
3ffffe90:  3ffe8962 3ffef600 3fffff40 402081f5  
3ffffea0:  40214008 00000000 000003e8 402083b8
3ffffeb0:  3ffef600 3fffff18 3fffff40 40208305  
3ffffec0:  3ffef600 00000000 00000000 00000000
3ffffed0:  00000001 00000000 3ffef50c 40213a21  
3ffffee0:  3ffe8aa5 3ffe8962 3ffef50c 40201466
3ffffef0:  feefeffe feefeffe feefeffe feefeffe
3fffff00:  feefeffe feefeffe feefeffe feefeffe  
3fffff10:  feefeffe feefeffe feefeffe feefeffe  
3fffff20:  feefeffe feefeffe feefeffe feefeffe
3fffff30:  feefeffe feefeffe feefeffe feefeffe
3fffff40:  40214008 00000000 000003e8 402083b8
3fffff50:  00000000 00000000 00000000 00000000  
3fffff60:  3ffef600 00000000 feefeffe feefeffe
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe feefeffe feefeffe
3fffff90:  feefeffe feefeffe feefeffe 3ffef588  
3fffffa0:  3fffdad0 00000000 3ffef548 4020998c
3fffffb0:  feefeffe feefeffe 3ffe85d0 40100d51  
<<<stack<<<
`

valeros, 2020-05-04:
Hi @rserroyen ! Does it work with Arduino IDE?

rserroyen, 2020-05-04:
No

valeros, 2020-05-04:
If it doesn't work with Arduino IDE, please open an issue in the framework repository  https://github.com/esp8266/Arduino/issues

closed 2020-05-04 by rserroyen