platformio/platform-espressif8266 #210: Platformio erase flash during upload

ivelichkevich, 2020-05-21:
PlatformIO, version 4.3.3
CLion 2020.1.1 Build #CL-201.7223.86, built on April 29, 2020
macOS 10.14.6

**Every time when I upload flash (fs) erased.**

```
platformio run -t upload -v                                                                                                                            
Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino; upload_port: /dev/cu.wchusbserial14210; upload_speed: 115200; monitor_port: /dev/cu.wchusbserial14210; monitor_speed: 115200; lib_deps: DHT sensor library for ESPx, ArduinoJson@<6.0, IRremoteESP8266, PubSubClient, rc-switch)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 2.5.1 > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash

"/usr/local/Cellar/platformio/4.3.3/libexec/bin/python3.8" "/Users/ihar.velichkevich/.platformio/packages/tool-esptoolpy/esptool.py" --before default_reset --after hard_reset --chip esp8266 --port "/dev/cu.wchusbserial14210" --baud 115200 write_flash 0x0 .pio/build/esp12e/firmware.bin
```

**There is no --erase-all option cmd line, but flash is empty every time.**

closed 2020-05-21 by ivelichkevich

atsju, 2022-06-05:
@ivelichkevich hi, I'm wondering, why did you close the issue ?
I find strange also that there is not target for erase flash like for ESP32 https://docs.platformio.org/en/latest/platforms/espressif32.html#erase-flash

pvpatel9726, 2023-02-28:
> @ivelichkevich hi, I'm wondering, why did you close the issue ? I find strange also that there is not target for erase flash like for ESP32 https://docs.platformio.org/en/latest/platforms/espressif32.html#erase-flash

 I also facing same issue for esp8266 in platformIO.  pltformIO erase the all flash memory on every upload. Is there any option like in arduino IDE in tools>Erase Flash> {Sketch, Sketch+WiFi, All Flash Content} . ?

[image]