platformio/platform-espressif8266 #215: Option `board_build.f_flash` is not used

Jason2866, 2020-05-30:
Using this setting does work for setting CPU frequency to 160Mhz
but the setting for flash frequency is not used.
```

[common]
board_build.f_cpu         = 160000000L
board_build.f_flash       = 20000000L

```

snipplet from Terminal verbose Output:

```
 "c:\python37\python.exe" "C:\Users\Hans\.platformio\packages\framework-arduinoespressif8266\tools\elf2bin.py" --eboot "C:\Users\Hans\.platformio\packages\framework-arduinoespressif8266\bootloaders\eboot\eboot.elf" --app .pioenvs\tasmota-universal\firmware.elf --flash_mode dout --flash_freq 40 --flash_size 1M --path "C:\Users\Hans\.platformio\packages\toolchain-xtensa\bin" --out .pioenvs\tasmota-universal\firmware.bin
```
There is still the default setting `--flash_freq 40`

Used latest PlatformIO develeopment version

valeros, 2020-06-01:
Hi @Jason2866 ! Please attach here the entire `platformio.ini` file.

Jason2866, 2020-06-01:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/en/stable/projectconf.html

[platformio]
description = Provide ESP8266 based devices with Web, MQTT and OTA firmware
src_dir = tasmota
build_dir = .pioenvs
workspace_dir = .pioenvs
build_cache_dir = .cache
extra_configs = platformio_tasmota_env.ini
                platformio_override.ini

; *** Build/upload environment
default_envs =
; *** Uncomment by deleting ";" in the line(s) below to select version(s)
;                tasmota
;                tasmota-ircustom
;                tasmota-minimal
;                tasmota-lite
;                tasmota-knx
;                tasmota-sensors
;                tasmota-display
;                tasmota-ir
;                tasmota-BG
;                tasmota-BR
;                tasmota-CN
;                tasmota-CZ
;                tasmota-DE
;                tasmota-ES
;                tasmota-FR
;                tasmota-GR
;                tasmota-HE
;                tasmota-HU
;                tasmota-IT
;                tasmota-KO
;                tasmota-NL
;                tasmota-PL
;                tasmota-PT
;                tasmota-RO
;                tasmota-RU
;                tasmota-SE
;                tasmota-SK
;                tasmota-TR
;                tasmota-TW
;                tasmota-UK
;
; *** alternatively can be done in: platformio_override.ini
; *** See example: platformio_override_sample.ini
; *********************************************************************

[common]
framework                 = arduino
board                     = esp01_1m
board_build.flash_mode    = dout
board_build.f_cpu         = 160000000L
board_build.f_flash       = 20000000L
board_build.ldscript      = eagle.flash.1m.ld

platform                  = ${core_active.platform}
platform_packages         = ${core_active.platform_packages}
build_flags               = ${core_active.build_flags}

; *********************************************************************
; *** Use custom settings from file user_config_override.h
                            -DUSE_CONFIG_OVERRIDE
; *********************************************************************

; *** Fix espressif8266@1.7.0 induced undesired all warnings
build_unflags             = -Wall

board_build.f_cpu         = 80000000L
monitor_speed             = 115200
upload_speed              = 115200
; *** Upload Serial reset method for Wemos and NodeMCU
upload_resetmethod        = nodemcu
upload_port               = COM5
extra_scripts             = ${scripts_defaults.extra_scripts}

[scripts_defaults]
extra_scripts             = pio/strip-floats.py
                            pio/name-firmware.py
                            pio/gzip-firmware.py
                            pio/override_copy.py

[esp_defaults]
build_flags               = -D_IR_ENABLE_DEFAULT_=false
                            -DDECODE_HASH=true -DDECODE_NEC=true -DSEND_NEC=true
                            -DDECODE_RC5=true -DSEND_RC5=true -DDECODE_RC6=true -DSEND_RC6=true
; new mechanism to set the IRremoteESP8266 supported protocols: none except HASH, NEC, RC5, RC6

[esp82xx_defaults]
build_flags               = ${esp_defaults.build_flags}
                            -Wl,-Map,firmware.map
                            -D NDEBUG
                            -mtarget-align
                            -DFP_IN_IROM
                            -DBEARSSL_SSL_BASIC
                            ; NONOSDK22x_190703 = 2.2.2-dev(38a443e)
                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703
                            ; lwIP 2 - Higher Bandwidth no Features
                            -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH
                            ; VTABLES in Flash
                            -DVTABLES_IN_FLASH
                            ; No exception code in firmware
                            -fno-exceptions
                            -lstdc++                            
                            ; the following removes the 4-bytes alignment for PSTR(), waiting for a cleaner flag from Arduino Core
                            -DPSTR\(s\)=\(__extension__\(\{static\ const\ char\ __c\[\]\ __attribute__\(\(__aligned__\(1\)\)\)\ __attribute__\(\(section\(\ \"\\\\\".irom0.pstr.\"\ __FILE__\ \".\"\ __STRINGIZE\(__LINE__\)\ \".\"\ \ __STRINGIZE\(__COUNTER__\)\ \"\\\\\"\,\ \\\\\"aSM\\\\\"\,\ \@progbits\,\ 1\ \#\"\)\)\)\ =\ \(s\)\;\ \&__c\[0\]\;\}\)\)
                            -DPSTR_ALIGN=1
                            ; restrict to minimal mime-types
                            -DMIMETYPE_MINIMAL

[irremoteesp_full]
build_flags               = -DUSE_IR_REMOTE_FULL
                            -U_IR_ENABLE_DEFAULT_
                            -DDECODE_PRONTO=false -DSEND_PRONTO=false

[core_active]
platform                  = ${tasmota_core.platform}
platform_packages         = ${tasmota_core.platform_packages}
build_flags               = ${tasmota_core.build_flags}

[tasmota_core]
; *** Esp8266 Arduino core 2.7.1
platform                  = espressif8266@2.5.1
platform_packages         = 
build_flags               = ${esp82xx_defaults.build_flags}
```

valeros, 2020-06-01:
I don't any usages of `common` section in your `platformio.ini`. You need to explicitly refer to this field in your specific env:
```
platform = espressif8266
framework = arduino
board_build.f_flash = ${common.board_build.f_flash}
```

Jason2866, 2020-06-01:
Ahh💥 thank you. :man_facepalming:

closed 2020-06-01 by Jason2866