platformio/platform-espressif8266 #217: OTA PlatformIO "Sending invitation to:  [ERROR]: No Answer"

rserroyen, 2020-06-09:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					Wemos d1 mini PRO 16MB 
- Core Version:			[latest git hash or date]
- Development Env:		PlatformIO/Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 R1 Mini Pro
- Upload Using:		espota

### Problem Description

Hello

When I do the Sketch in Arduino you are fine!

When I execute the Sketch in PlatformIO "Sending invitation to: [ERROR]: No Answer"

Why ??

Thank you

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
[env:d1_mini_pro]
platform = espressif8266
board = d1_mini_pro
framework = arduino

;upload_port = COM8
monitor_speed = 115200

upload_port = 192.168.1.53
upload_protocol = espota
upload_flags = --port=3232
------------------------------------------------------------

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

#define mySSID "WiFi-2.4-4CE9"; // WiFi SSID
#define myPASSWORD "852CEDEC6D"; // WiFi PASSWORD
#define myHostname  "esp8266"; // Hostname

const char* ssid = mySSID;
const char* password = myPASSWORD;
const char* Hostname = myHostname;

void setup() {
    pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 8266
  // ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  ArduinoOTA.setHostname(Hostname);

  // No authentication by default
  ArduinoOTA.setPassword((const char *)"123");

  ArduinoOTA.onStart([]() {
    Serial.println("Start");
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();
  digitalWrite(LED_BUILTIN, LOW);
    delay(100);
    digitalWrite(LED_BUILTIN, HIGH);
    delay(100);
}
```

### Debug Messages

``
Executable segment sizes:
IROM   : 281600          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27864   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1280  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 1204  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 25464 )         - zeroed variables      (global, static) in RAM/HEAP 

Sketch uses 311948 bytes (29%) of program storage space. Maximum is 1044464 bytes.
Global variables use 27948 bytes (34%) of dynamic memory, leaving 53972 bytes for local variables. Maximum is 81920 bytes.
Authenticating...OK
Uploading..................................................................................................................................... OK

Uploading .pio\build\d1_mini_pro\firmware.bin
10:40:02 [DEBUG]: Options: {'esp_ip': '192.168.1.53', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 58911, 'auth': '', 'image': '.pio\\build\\dbin', 'spiffs': False, 'debug': True, 'progress': True}
10:40:02 [INFO]: Starting on 0.0.0.0:58911
10:40:02 [INFO]: Upload size: 316192
10:40:02 [INFO]: Sending invitation to: 192.168.1.53
10:40:12 [ERROR]: No Answer
*** [upload] Error 1
======================================================================= [FAILED] Took 33.47 seconds ==================================================
The terminal process terminated with exit code: 1
```

Pablo2048, 2020-06-09:
Hi,
probably it's because wrong port - 3232 is OTA port for ESP32 device AFAIK. Try to remove `upload_flags=...` line from your platformio.ini and try again.

rserroyen, 2020-06-09:
Hi

Without upload_flags = --port=3232 ------ same 
`13:27:29 [INFO]: Upload size: 316224
13:27:29 [INFO]: Sending invitation to: 192.168.1.53
13:27:39 [ERROR]: No Answer
*** [upload] Error 1`

Pablo2048, 2020-06-09:
Does your esp8266 respond to `ping 192.168.1.53`? Does it respond to `ping esp8266.local`? Also you can try if it shows in PIO Home->Devices->Multicast DNS.

rserroyen, 2020-06-10:
Change ip to IP address: 192.168.1.59

Multicast DNS > Refresh
No host !!!

But work !!!

```
ping 192.168.1.59

Envoi d’une requête 'Ping'  192.168.1.59 avec 32 octets de données :
Réponse de 192.168.1.59 : octets=32 temps=13 ms TTL=255
Réponse de 192.168.1.59 : octets=32 temps=24 ms TTL=255
Réponse de 192.168.1.59 : octets=32 temps=35 ms TTL=255
Réponse de 192.168.1.59 : octets=32 temps=50 ms TTL=255

Statistiques Ping pour 192.168.1.59:
    Paquets : envoyés = 4, reçus = 4, perdus = 0 (perte 0%),
Durée approximative des boucles en millisecondes :
    Minimum = 13ms, Maximum = 50ms, Moyenne = 30ms

ping esp8266.local

Envoi d’une requête 'ping' sur esp8266.local [192.168.1.59] avec 32 octets de données :
Réponse de 192.168.1.59 : octets=32 temps=3 ms TTL=255
Réponse de 192.168.1.59 : octets=32 temps=20 ms TTL=255
Réponse de 192.168.1.59 : octets=32 temps=31 ms TTL=255
Réponse de 192.168.1.59 : octets=32 temps=48 ms TTL=255

Statistiques Ping pour 192.168.1.59:
    Paquets : envoyés = 4, reçus = 4, perdus = 0 (perte 0%),
Durée approximative des boucles en millisecondes :
    Minimum = 3ms, Maximum = 48ms, Moyenne = 25ms

ping esp8266.local

Envoi d’une requête 'ping' sur esp8266.local [192.168.1.59] avec 32 octets de données :
Réponse de 192.168.1.59 : octets=32 temps=3 ms TTL=255
Réponse de 192.168.1.59 : octets=32 temps=20 ms TTL=255
Réponse de 192.168.1.59 : octets=32 temps=31 ms TTL=255
Réponse de 192.168.1.59 : octets=32 temps=48 ms TTL=255

Statistiques Ping pour 192.168.1.59:
    Paquets : envoyés = 4, reçus = 4, perdus = 0 (perte 0%),
Durée approximative des boucles en millisecondes :
    Minimum = 3ms, Maximum = 48ms, Moyenne = 25ms

```