platformio/platform-espressif8266 #219: Update ESP8266_RTOS_SDK v1.4.0 to ESP8266_RTOS_SDK v3.x.x

rajeshpachaikani, 2020-07-08:
The version being used is awfully old and no documentation available for it on Espressif's website.

BluRaf, 2020-11-24:
This problem still exists, I had to migrate project to standard Espressif RTOS SDK because of outdated version used in PlatformIO.

Seramis, 2021-04-06:
Yup, i write here to give support for updating the SDK for ESP8266.

Seramis, 2021-04-06:
#85 
#125 
#192 

So, any plans?

pamafe1976, 2021-04-09:
I am also interested in this update

yooshky, 2021-04-13:
ESP8266_RTOS_SDK v1.4.0 is unuseable, PIO for me too.

shreekant-sureliya, 2021-05-01:
is there any update...?

utkarshsethi, 2021-05-06:
@ivankravets @valeros please help with this

randaljunior, 2021-11-15:
Is It too hard to make this jump?
no one on PlatformIO cares?

mtlcrusher, 2021-12-20:
I would like to help, however, I don't know where is the repos that supposed to be edited?

koluckirafal, 2021-12-20:
> I would like to help, however, I don't know where is the repos that supposed to be edited?

https://github.com/platformio/platform-espressif8266/

1. [platform.json](https://github.com/platformio/platform-espressif8266/blob/develop/platform.json#L58-L63) - here you need to bump version, currently release v3.4: https://github.com/espressif/ESP8266_RTOS_SDK/releases/tag/v3.4
2. [builder/frameworks/esp8266-rtos-sdk.py](https://github.com/platformio/platform-espressif8266/blob/develop/builder/frameworks/esp8266-rtos-sdk.py) - builder script needs to be updated for the RTOS SDK 3.x, which had some breaking changes to make it closer to the ESP-IDF framework. You might want to check how is ESP-IDF support implemented here: https://github.com/platformio/platform-espressif32/blob/develop/builder/frameworks/espidf.py

seancale, 2022-06-06:
Still looks like it's trying to use the same ancient SDK.

My platform.json has this, even for the development version:
```   
 "framework-esp8266-rtos-sdk": {
      "type": "framework",
      "optional": true,
      "owner": "platformio",
      "version": ">=1.5.0-beta"
    },```

I'm running PlatformIO Core version 6.0.2 on VS Code 1.67.2. Device is a generic ESP8285.

bwjohns4, 2022-12-09:
Any update on this?

r315, 2023-06-03:
Does platform.io will ever support ESP8266_RTOS_SDK v3.x ?

hernot, 2023-10-08:
There is a pending pull-request #284 from @freedib which would benefit from some more up-votes and commets from eagerly awaiting users.

jeecrypt, 2023-10-13:
I would like to return to PlatformIO, but the old version of RTOS SDK does not allow me to do this. I'm really looking forward to the update. I want stability.

bebek, 2024-02-15:
Is there any commitment to do the SDK upgrade or are we facing a dead end?
We cannot use anymore PIO for the esp8266 with RTOS.

g0eki, 2025-04-08:
I am also interested and very grateful.

hernot, 2025-06-08:
ba the way issue #125 seems to request the same.

freedib, 2025-06-08:
Interesting issue. When I implemented IDF style RTOS, I removed support for 1.5 to avoid naming confusion.
I don't understand why Espressif did not name the new SDK: ESP8266-IDF.
It could easily bring it back. 
I suggest to keep the framework name "esp8266-rtos-sdk" for RTOS 3.4 to be coherent with Espressif  and use something like "esp8266-rtos15-sdk" or "esp8266-rtos-legacy-sdk" for RTOS 1.5

freedib, 2025-06-09:
You could create a partition.csv file for 512k to get a warning at compile time, but I suspect that components libraries in IDF style are more complex and bigger than legacy ones, and as experimented, except basic programs it won't fit in 512k.

hernot, 2025-06-09:
> Interesting issue. When I implemented IDF style RTOS, I removed support for 1.5 to avoid naming confusion. I don't understand why Espressif did not name the new SDK: ESP8266-IDF. It could easily bring it back.

A wild blind guess. The esp8266 3.x rtos sdk is meant as intermediate release until native esp8266 ESP-IDF support is finished and thus has nothing in common with the latter apart from API compatibility. 

 
> I suggest to keep the framework name "esp8266-rtos-sdk" for RTOS 3.4 to be coherent with Espressif and use something like "esp8266-rtos15-sdk" or "esp8266-rtos-legacy-sdk" for RTOS 1.5

I fully do support your suggestion.

> You could create a partition.csv file for 512k to get a warning at compile time, but I suspect that components libraries in IDF style are more complex and bigger than legacy ones, and as experimented, except basic programs it won't fit in 512k.

Just a quick question is there any pio or Espressif tool to extract partition.csv from `eagle.app.v6.ld` linker script or what is the safest approach to do it manually? Which pitfalls and human fails are to be considered when doing manually?

freedib, 2025-06-09:
I don't think such a tool exists and I never modified a partition table but according to https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/api-guides/partition-tables.html I would try the following partition.csv:

# Name,   Type, SubType, Offset,  Size
nvs,      data, nvs,     0x9000,  0x3000
factory,  app,  factory, 0xC000, 0x74000

ESP8266 is almost discontinued and replaced by ESP32-C3. The platform upgrade I proposed 2 years ago made sense at that time, but now, I suspect Platformio people won't put energy in supporting  a new platform for a product which is in end of life. So I guess RTOS 1.5 will stay the standard on Platformio.

hernot, 2025-06-09:
> I don't think such a tool exists and I never modified a partition table but according to https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/api-guides/partition-tables.html I would try [...]

Good argument. So how much efforts should be put into a `partition.csv` for low flash ( <1M ) chips and board anyway? Would be anything beyond a NOTE/WARNING in the README.md in your esp8266 rtos sdk 3.x Repo and related platform  repos worth the efforts at all?

freedib, 2025-06-10:
I have no Sparkfun Thing to test but the proposed partitions.csv seems to work with standard idf.py. 
With platformio and my proposed esp8266 platform, there is a bug with partition file I would have to correct.

Note that even with optimized partition, as soon as you put TCP/IP in code the code gets too big due to RTOS 3.4 new IP stack.
So if you want to stay with Sparkfun Thing there is no solution. Stay with RTOS 1.5 or buy a 5$ 16 Mbytes D1-mini on Ali Express!