platformio/platform-espressif8266 #220: 2.6.0 platform does not install mklittlefs package

CODeRUS, 2020-07-10:
Hello!

Just switched to 2.6.0 to test new littlefs functionality, but buildfs fails

```
pio run --target buildfs -e nodemcu                                                                        :(
Processing nodemcu (framework: arduino; platform: espressif8266@2.6.0; board: nodemcuv2)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 2.6.0 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
- framework-arduinoespressif8266 3.20702.0 (2.7.2)
- tool-esptool 1.413.0 (4.13)
- tool-esptoolpy 1.20800.0 (2.8.0)
- tool-mkspiffs 1.200.0 (2.0)
- toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESPAsyncE131-wled>
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncUDP> 0.0.0-alpha+sha.697c75a025
|-- <ESPReactWifiManager> 0.1 #9aba504
|   |-- <ESP Async WebServer> 1.2.3 #f13685e
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ArduinoJson> 6.15.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 6.15.2
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|-- <arduinoFFT> 1.5.5
|-- <NTPClient> 3.1.0
|-- <ESP8266WiFi> 1.0
|-- <ArduinoJson> 6.15.2
|-- <FastLED> 3.3.3
|   |-- <SPI> 1.0
|   |-- <EspSoftwareSerial> 6.8.5
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <FastLED NeoMatrix> 1.2 #be7f353
|   |-- <Framebuffer GFX> 1.0 #8eff8f0
|   |   |-- <FastLED> 3.3.3
|   |   |   |-- <SPI> 1.0
|   |   |   |-- <EspSoftwareSerial> 6.8.5
|   |   |-- <Adafruit GFX Library> 1.9.0
|   |   |   |-- <Adafruit BusIO> 1.3.3
|   |   |   |   |-- <SPI> 1.0
|   |   |   |-- <SPI> 1.0
|   |-- <Adafruit GFX Library> 1.9.0
|   |   |-- <Adafruit BusIO> 1.3.3
|   |   |   |-- <SPI> 1.0
|   |   |-- <SPI> 1.0
|   |-- <FastLED> 3.3.3
|   |   |-- <SPI> 1.0
|   |   |-- <EspSoftwareSerial> 6.8.5
|-- <ESP Async WebServer> 1.2.3 #f13685e
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ArduinoJson> 6.15.2
|   |-- <ESP8266WiFi> 1.0
|-- <GyverButton>
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
Building in release mode
Building file system image from 'data' directory to .pio/build/nodemcu/littlefs.bin
sh: 1: mklittlefs: not found
*** [.pio/build/nodemcu/littlefs.bin] Error 127
```

platformio.ini:
```
[env:nodemcu]
framework                 = arduino
platform                  = espressif8266@2.6.0
board                     = nodemcuv2
board_build.filesystem    = littlefs

build_flags               = ${common.build_flags}

monitor_speed             = ${common.monitor_speed}
upload_speed              = ${common.upload_speed}
```

CODeRUS, 2020-07-10:
Any way to install tool-mklittlefs manually?

valeros, 2020-07-10:
Hi @CODeRUS ! Please run `pio system info` in the IDE terminal and attach here the output.

CODeRUS, 2020-07-10:
weird

```
pio system info                                                                                            :(
Usage: pio system [OPTIONS] COMMAND [ARGS]...

Error: No such command "info".
```

valeros, 2020-07-10:
OK, what is your OS, architecture?

CODeRUS, 2020-07-10:
ubuntu 20.04
pio was installed with vscode, long time ago, i think...
```
pio --version                                                                                              :(
PlatformIO, version 4.3.4
```

valeros, 2020-07-10:
Please run `uname -a` in the terminal.
Also, do you see the folder `tool-mklittlefs` in `/home/YOUR_USER/.platformio/packages`? Are there any files in this folder?

CODeRUS, 2020-07-10:
`Linux desktop 5.4.0-40-generic #44-Ubuntu SMP Tue Jun 23 00:01:04 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

`find ~/.platformio | grep mklittlefs` says nothing

valeros, 2020-07-10:
Could you please try to run `pio run --target uploadfs -e nodemcu` ?

CODeRUS, 2020-07-10:
after  `uplaodfs` mklittlefs tool was installed, thanks!

valeros, 2020-07-10:
Resolved in https://github.com/platformio/platform-espressif8266/commit/b27329b6cfe26247a976497cd10d1e8e971dbd9c

closed 2020-07-10 by valeros

johnsmakar, 2021-05-23:
I have a question, I am able to upload my data correctly but want to confirm that it is using LittleFS and not SPIFFS.
```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/tinypico.html
PLATFORM: Espressif 32 (3.2.0) > TinyPICO
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LittleFS_esp32> 1.0.6
|   |-- <FS> 1.0
|-- <FS> 1.0
Building in release mode
Replace MKSPIFFSTOOL with mklittlefs.exe
Building SPIFFS image from 'data' directory to .pio\build\tinypico\spiffs.bin
/config/calibration.txt
/log.csv
Looking for upload port...
Auto-detected: COM7
Uploading .pio\build\tinypico\spiffs.bin
esptool.py v3.0
Serial port COM7
Connecting......
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:b9:7e:82:4b:a4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1507328 bytes to 1966...
Writing at 0x00290000... (100 %)
Wrote 1507328 bytes (1966 compressed) at 0x00290000 in 0.0 seconds (effective 246751.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
I see in packages it shows `tool-mkspiffs 2.230.0 (2.30)` is that fine for littleFS or is there a "tool-mklittlefs" package I need to install?

It also says `Building SPIFFS image from 'data' directory to .pio\build\tinypico\spiffs.bin` which has me questioning if I am doing this correct.

valeros, 2021-05-24:
@johnsmakar You can run the upload command in verbose mode to see what tools are used. Here is some [docs](https://docs.platformio.org/en/latest/platforms/espressif8266.html#selecting-appropriate-filesystem) describing how to select the filesystem.

johnsmakar, 2021-05-24:
I have `board_build.filesystem = littlefs ` in my .ini but I have a feeling it's not doing anything because when I make a typo on purpose it does not complain.

valeros, 2021-05-24:
@johnsmakar It seems that you're using ESP32-based hardware, but writing in the ESP8266 repository. Anyway, the ESP32 platform doesn't support `littleFS` at the moment.

johnsmakar, 2021-05-24:
Ah, my apologies. I am slightly confused on what filesystem my TinyPICO is running then, because I am using the [ESP32 LittleFS filesystem plugin](https://github.com/lorol/arduino-esp32fs-plugin) to upload my FS image to the board. 

And I saw this and assumed it was supported: https://www.youtube.com/watch?v=4r6YZlLfKfw&list=LL&index=3&t=1112s