platformio/platform-espressif8266 #221: ESP32 and ESP8266 EEPROM

hzhh110, 2020-07-14:
ESP8266 When the EEPROM size exceeds 1000, the firmware upgrade fails, but ESP32 does not have this situation. Isn't EEPROM 4096?

hzhh110, 2020-07-14:
[env:esp12e]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps  = AliyunMqttArduino,ArduinoJson,NTPClient
build_flags =
  -D MQTT_MAX_PACKET_SIZE=1024
  -D MQTT_KEEPALIVE=60

ESP8266 When the EEPROM size exceeds 1000, it is easy to trigger a restart,The ESP32 does not have this phenomenon

115200:
ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld

9600:
ser exception (panic/abort/assert)
Panic core_esp8266_main.cpp:133 __yield

>>>stack>>>

ctx: sys
sp: 3fffff10 end: 3fffffb0 offset: 0000
3fffff10:  00008f37 3ffef80c 3ffef40c 40210558  
3fffff20:  000000fe 00000000 00000000 00000000  
3fffff30:  00000000 00000000 00000000 3ffefbc8  
3fffff40:  3ffefbe8 3ffefbe8 3ffefbe8 3ffefb88  
3fffff50:  3fffdad0 00000000 3ffef894 4020e6b6  
3fffff60:  00000000 00b10dd2 4020e794 4020e713  
3fffff70:  4020e7ad 00000014 3ffefb88 4020e1e9  
3fffff80:  00000000 00000000 3ffef894 40205eba  
3fffff90:  3fffdad0 00000000 3ffefb48 401001f1  
3fffffa0:  feefeffe 00000000 3ffefb48 4020e2bc  
<<<stack<<<

valeros, 2020-07-14:
Hi @hzhh110 ! Does it work normally in Arduino IDE?

hzhh110, 2020-07-15:
@valeros I have been using PlatformIO of vscode and did not compile with Arduino IDE

hzhh110, 2020-07-15:
> Hi @hzhh110 ! Does it work normally in Arduino IDE?

 I have been using PlatformIO of vscode and did not compile with Arduino IDE

valeros, 2020-07-15:
Would be great if you could try the same code with Arduino IDE.